m-labs/artiq #14: logging formatting styles cleanup

jordens, 2015-03-25:
The logging message formatting is a inconsistent. Let's do the following:

* `style="{"` should be added to https://github.com/m-labs/artiq/blob/master/artiq/tools.py#L61
* https://github.com/m-labs/artiq/blob/master/artiq/coredevice/comm_serial.py should be cleaned up to use `logger.level("format {}", *arguments)` instead of `logger.level("format {}".format(*arguments))`. The latter is always evaluated which the former is evaluated only on message emission.
* all the other uses of logging should be switch from `%` to `{}` formatting.

sbourdeauducq, 2015-03-26:
Did you get ``style="{"`` to work? I didn't. I also worry it may break compatibility with external modules (e.g. ``asyncio``) that use old-style formatting.

sbourdeauducq, 2015-03-26:
This does not work:

```python
import logging

logger = logging.getLogger('simple_example')
ch = logging.StreamHandler()
formatter = logging.Formatter(style="{")
ch.setFormatter(formatter)
logger.addHandler(ch)

logger.warn('warn message {0}', 42)
```

jordens, 2015-03-26:
Ack. Looks broken. Then let's keep the "%" formatting for logging which then only requires changing `comm_serial.py` to use lazy formatting.

closed 2015-03-27 by sbourdeauducq