m-labs/artiq #17: flash_storage issues

sbourdeauducq, 2015-04-27:
* the method of access to data fields is inconsistent across functions (``union seek``,  ``*(unsigned int *)addr == END_MARKER``, etc.). I guess you can remove ``union seek``.
* ``write_at_offset`` may access buffers beyond their length due to the added alignment padding. Likely harmless, but not a clean programming practice.
* ``fs_write`` will break if the data contains the FFFFFFFF pattern. Forgot to use ``goto_next_record``?
* you can probably remove some of the casts and clutter by trying to use ``int`` (instead of ``unsigned int``) everywhere.
* ``strlen`` may crash if a record is corrupted. Use ``strnlen``. Generally make the code resistant to broken data.
* variable names are inconsistent (e.g. ``buflen`` vs. ``key_len``)
* ``fs_write`` should return a status code indicating if the data was successfully written or not.

NB: All those bugs would have been prevented by using YAFFS.

sbourdeauducq, 2015-04-30:
You can get rid of the alignment requirements by doing:
int o;
memcpy(&o, source, 4);

sbourdeauducq, 2015-04-30:
Also you should put the size first in the record, as a size of 0xffffffff is unlikely whereas a string starting with 0xffffffff is more valid. Don't forget to update artiq_mkfs.

sbourdeauducq, 2015-05-01:
How does one delete a record? Write it with zero length data?

fallen, 2015-05-10:
I addressed the issues in this commit that you can cherry-pick on master (it should apply): https://github.com/fallen/artiq/commit/2ac8c8c7b84780c10cafd056b561342c5a9183e1

* I added some tests (fstest command in test_mode.c prompt) to be able to more easily avoid regressions in the future
* I used strnlen when possible
* I only put the size of the total record at the beginning of the record
* To delete a record you indeed write it with zero length data, this kind of duplicate is cleaned when sector is full and you try to write to it, there is fs_remove() macro to do the key removing.
* There are no alignment issues anymore
* I updated the artiq_mkfs tool
* You can now detect easily if a key exists while reading it or if flash storage is full when doing a write (and therefore the write fails)
  * fs_write() returns a non-zero (1) value if flash storage is full, returns 0 if everything went OK
  * fs_read() returns 0 (as read length) if the key does not exist

fallen, 2015-05-10:
Thanks for the review and sorry for the delay in fixing those points.

sbourdeauducq, 2015-05-11:
More issues: https://github.com/fallen/artiq/commit/2ac8c8c7b84780c10cafd056b561342c5a9183e1

closed 2015-06-04 by sbourdeauducq