m-labs/artiq #18: Build for KC705 in Travis

sbourdeauducq, 2015-04-27:
Uniprocessor systems and therefore the Papilio Pro are no longer supported. Build for KC705 instead as this it is the main board to be used with ARTIQ.

jboulder, 2015-04-28:
If support for Papilio Pro is depricated it would be good to update various
parts of m-labs/artiq to reflect this. Some references I can find...
* http://m-labs.hk/artiq/manual/fpga_board_ports.html?highlight=papilio
* http://m-labs.hk/artiq/index.html
* http://m-labs.hk/artiq/manual/installing.html

Is Pipistrello now the officially supported low-end FPGA board? -Joe

sbourdeauducq, 2015-04-29:
Updated, thanks for the reminder. Yann, if we are going to mention pipistrello binaries on the website, we need a binary package for it too (if building both binaries is a mess, then update the website).

fallen, 2015-05-10:
If I understand well what needs to be done:
* Remove the building of the papilio pro bitstream+bios+runtime from the conda recipe as papilio pro isn't supported anymore
* Build pipistrello bitstream+bios+runtime in conda recipe and embed those binaries in the conda package (like it was done with papilio pro before)
* Try to build also for KC705 and embed the KC705 binaries in the conda packages

If the pipistrello build works, then I update the website to just say "install the conda package" + "flash via artiq_flash"
If the pipistrello build does not work, then I update the binary_package.tar.gz file to contain real pipistrello binaries and let the text as is

Is that alright?

jordens, 2015-05-11:
LGTM
If building for KC705 in CI keeps being difficult, I would consider starting to do something like releases now. If it all works mostly (before show-and-tells, tutorials, etc), tag an artiq commit and build bitstream+bios+runtime+conda package for it and upload.

fallen, 2015-05-29:
As statedin the status report of May, both KC705 and Pipistrello bitstreams are now built by Travis, moreover the bitstreams+bios+runtimes are included in the conda packages so that any newcomer can just:
$ conda install artiq
$ artiq_flash.sh

And it should work out of the box.
Before closing this ticket, I still need to update the website : 
* to say to flash via artiq_flash instead of downloading the binary package
* add missing conda config --add channels http://conda.binstar.org/fallen/channel/dev

fallen, 2015-06-15:
This is fixed now:
* The manual contains instructions on how to either install from conda or sources, saying conda is the preferred way.
* linux-32 and linux-64 packages contain prebuilt bitstreams/BIOS/runtime
* http://m-labs.hk/artiq install instructions now point directly to ARTIQ manual
* "main" conda channel is not needed anymore if you want to follow the dev, just use the "dev" channel which contains *all* needed dependencies. No risk of conflicts between the two channels anymore.

closed 2015-06-15 by fallen