m-labs/artiq #23: lda controller segfaults when device cannot be opened

sbourdeauducq, 2015-06-18:


fallen, 2015-06-19:
After a bit of investigation, it seems like a hidapi bug.
I can reproduce it even by using the hidapi-libusb library from C, see over there: https://gist.github.com/fallen/b8af26da2e98e51a99b5
Also, I just found it is listed in the library's issues: https://github.com/signal11/hidapi/issues/185

We could either try to work around this bug, for instance by always passing NULL as hid_open's 3rd argument. But that would mean we would not be able to use several devices on the same machine.

Or, we can just fork hidapi, add the extra check for NULL, and package it in a conda package.

I will try to send a patch to the author, maybe he will merge it quickly...

sbourdeauducq, 2015-06-19:
I think we need to keep the possibility to use several devices on the same machine.

sbourdeauducq, 2015-06-19:
Shipping hidapi for all platforms sounds like a mess and it will potentially conflict with system versions. Let's not work around this in ARTIQ - it's minor and the proper fix is to release/package the lib through the upstream channels.

sbourdeauducq, 2019-04-19:
Fixed for Nix users, others should live with the problem or compile hidapi from source using the latest git version.

closed 2019-04-19 by sbourdeauducq