m-labs/artiq #24: lda driver supports using the first enumerated device, but controller does not expose this option

sbourdeauducq, 2015-06-18:


fallen, 2015-06-26:
We could make use of the device enumeration by passing ``None`` to the driver in case ``-d`` is not used (it already defaults to ``None``).
For simulation: we could have ``-P/--product`` default to ``None`` and have this mean "simulation mode".

The downside of this is that other controllers (3 of them) enter simulation mode when omitting ``-d/--device``.
So we lose a bit of consistency if we use ``-P/--product`` for that in the lda controller :/
On the other hand, the pxi6733 controller is using ``-C/--channels`` for this purpose, however it does not have any ``-d/--device`` parameter.

Or we could just drop the "using first enumerated device" from the driver.

jordens, 2015-06-26:
Having "no product" mean simulation mode is a bit cryptic for me. I would suggest requiring the serial number always. Explicit is nice. The fact that the lda driver can use the first enumerated device does not really need to be exposed on the controller. Regarding simulation mode, they should be a different commandline switch alltogether. Then all the problems due to abusing other switches to trigger simulation mode are not an issue anymore.

fallen, 2015-06-26:
All right, adding an explicit ``--simulation`` to all controllers then?
I'm OK with that.

jordens, 2015-06-27:
Yeah.

sbourdeauducq, 2015-06-27:
Sounds good.

fallen, 2015-06-29:
Implemented in https://github.com/m-labs/artiq/commit/515aa968194f7e7578ebec518d5a41371cef341e

sbourdeauducq, 2015-06-29:
AFAICT after this change some controllers will misbehave if you omit -d.

fallen, 2015-06-29:
For LDA it's not an issue but indeed for other controllers the following scenario will cause an error:
* omitting both ``--simulation`` and ``-d/--device`` (or both ``--simulation`` and ``-C/--channels`` for pxi6733 case)

I could just test for that with some:
`if not args.simulation and args.device is None:`
which would print an error message instructing to use one of the two options and tell to use ``-h/--help`` for more information.

Or, 
In those problematic controllers replacing:
`if args.simulation:`
with
`if args.simulation or args.device is None:`
I would then need to add to the help message of ``-d/--device`` parameter that omitting this parameter implies simulation mode.

which one is preferred?

sbourdeauducq, 2015-06-29:
1st one I would say.

fallen, 2015-06-29:
Implemented in https://github.com/m-labs/artiq/commit/a73776bd7297cb8caf467be6ad28a3ec5c7b38f7

sbourdeauducq, 2015-07-01:
What happens if you specify both --simulation and -d?

fallen, 2015-07-01:
As long as you specify ``--simulation``, you are in simulation mode

sbourdeauducq, 2015-07-01:
That should be documented. What's the use scenario for that? Modifying the device database to add --simulation while keeping the serial number stored?

fallen, 2015-07-01:
Yes for instance, or we could advocate for a separate "simulation" device database, dunno what's best.

fallen, 2015-07-01:
ack for the documentation point btw.

closed 2015-07-01 by sbourdeauducq