m-labs/artiq #25: RPC call to Thorlabs driver will return before the commanded action is complete

jboulder, 2015-06-18:
move_home is blocking
move_jog(direction, async=False) has blocking option
move_absolute() does not have blocking option
move_relative() does not have blocking option

fallen, 2015-06-19:
All those are now blocking since https://github.com/m-labs/artiq/commit/4d077f50c6621ad771818f489a5dd611e320ea6d
The only motor-moving and non-blocking call is ``move_velocity()`` which just moves the motor in one direction until you call ``move_stop()``.

I let this issue open until you test it and ack it works as expected.

jordens, 2015-06-21:
Blocking as default is good.

As a note for the future, I can imagine a few situations where you want to start a move to a certain point and then abort or override it with a different target. With fire-and-forget I suspect you can not abort a blocking move. If those situations come up, I would propose adding either a state to the driver that determines blocking behavior or add the `async` keyword argument back back in.

sbourdeauducq, 2015-06-21:
Tested working.

closed 2015-06-21 by sbourdeauducq