m-labs/artiq #27: document how to assign stable /dev/ttyUSBX nodes to tcube devices

sbourdeauducq, 2015-06-18:


jordens, 2015-06-18:
Use hwgrep. Also works on Windows.

fallen, 2015-06-19:
I can see one can do serial.open_for_url("hwgrep://VID:PID") which should work under both Windows+Linux as stated in this doc http://pyserial.sourceforge.net/pyserial_api.html

But that would mean we don't support multiple TDC/TPZ plugged on the same machine?

If we just bind via udev rules we can use the USB serial number to generate /dev/ttyUSBXX names consistently, then we can just open them "normally" like device files.
But indeed this (udev binding) only works on Linux.

Or maybe you are suggesting to use hwgrep only in the case when the user does not supply a serial device name? And to still let the user provide a name like "COMn" or "/dev/ttyUSBxx" and do static name binding via udev on Linux?

sbourdeauducq, 2015-06-19:
We definitely need to support multiple devices on the same machine. Typical use is through one of those fancy USB hub + power supply bars that have several devices docked.

sbourdeauducq, 2015-06-19:
"possible to search for the description or hardware ID of a device, e.g. USB VID:PID ***or texts***."
What does the "text" say?

fallen, 2015-06-19:
having a look.

fallen, 2015-06-19:
Done in https://github.com/m-labs/artiq/commit/b8bb3d8ca7fc1fcf62f2255c5b6a76c3622f7526

closed 2015-06-19 by fallen