m-labs/artiq #29: m-labs.hk/artiq/manual/ search

jboulder, 2015-06-18:
The text "sync()" appears in the artiq manual but using the web-based search for "sync()" doesn't find it.

http://m-labs.hk/artiq/manual/core_drivers_reference.html?highlight=sync#artiq.coredevice.ttl.TTLInOut.sync

jordens, 2015-06-18:
This is not a text search but an indexed keyword search. Search for `sync`. Also, this would be a bug in sphinx, not in artiq.

closed 2015-06-19 by sbourdeauducq

jboulder, 2015-06-21:
From a usability perspective being able to search the documentation is important. Consider updating the search box instructions from "Enter search terms or a module, class or function name" to "Search keyword index. e.g. sync not sync()."

jordens, 2015-06-21:
The function name of `sync` is `sync`, not `sync()`. You are welcome to file your suggestion with the sphinx project if you feel it is a correct improvement.

jboulder, 2015-06-21:
Robert, Do you maintain the m-labs website?

jordens, 2015-06-21:
No. I know sphinx and am triaging your issue.