m-labs/artiq #30: RTIO exception parameters are broken

sbourdeauducq, 2015-06-18:
Stuck at 0, e.g.
artiq.coredevice.runtime_exceptions.RTIOUnderflow: at 0 ps on channel 0, violation 0 ps

sbourdeauducq, 2015-06-28:
The bug happens when exceptions that are raised with parameters by the runtime are caught by a ``try/finally`` construct in the compiled code. The ``finally`` block re-raises the exception through the runtime's ``exception_raise``, which clears the parameters. Fixing this properly needs making the compiler aware of exception parameters - not clearing the parameters when the compiled code raises an exception is not enough, e.g.

<pre>
try:
    ...
except SomeException:
    # This does need to clear the parameters
    # (and change the ID of the exception re-raised by finally) 
    raise SomeOtherException
finally:
    ...
</pre>

This bug is manifesting itself all the time at the moment, as the whole kernel code is encapsulated in a ``try/finally`` construct to handle RTIO handovers.

whitequark, 2015-08-09:
new-py2llvm does not exhibit this bug.

sbourdeauducq, 2015-08-19:
Please close when new-py2llvm is usable with the rest of ARTIQ.

whitequark, 2015-11-23:
Closing since new compiler was merged.

closed 2015-11-23 by whitequark