m-labs/artiq #33: Document how to get pyserial hwgrep magic strings

sbourdeauducq, 2015-06-19:
Put in FAQ:
import serial.tools.list_ports
import pprint
pprint.pprint(serial.tools.list_ports.comports())

jordens, 2015-06-21:
or just `python3 -m serial.tools.list_ports -v`

fallen, 2015-06-23:
Done in https://github.com/m-labs/artiq/commit/7821d0f7c8e862eb8999bb3fc299ca21ad59850d

sbourdeauducq, 2015-06-23:
The purpose of this is to use serial number, not /dev/ttyUSBX nodes.

fallen, 2015-06-23:
Ack, fixed in https://github.com/m-labs/artiq/commit/f0dddd9f391d1da7b660e85f3ff92d353f532508

sbourdeauducq, 2015-06-23:
Still not clear and users already know how to list their serial ports (ls /dev, dmesg, windows control panel, etc.). So the FAQ entry should be entitled "...determine the pyserial URL to attach to a device by its serial number".

fallen, 2015-06-26:
Ah yes you're right, it's not clear : fixed in https://github.com/m-labs/artiq/commit/c38110201942a6ed982c44e5ae3f600b42c64d02
Sorry about that.

closed 2015-06-26 by fallen