m-labs/artiq #35: feature request: DDS bus reset

jboulder, 2015-06-20:
Command to reset all DDSs on DDS bus without requirement to reset FPGA.

jordens, 2015-06-21:
Rationale?
I can only imagine needing to reset the DDSs (do you want the bus/phy logic or the DDS) if you hit a bug in ARTIQ.

jboulder, 2015-06-21:
Use scenario: Artiq user suspects flakey DDS/TTL bus is resulting in erroneous frequency/phase. Reset of bus can test this. 
* pull rest pin on DDS
* reprogram configuration register
* continue with whatever experiment is scheduled next

jordens, 2015-06-21:
Sounds like the tools that are planned and already there are much better at confirming/excluding flakey DDS/TTL hardware and bugs in ARTIQ than a DDS reset would ever be.

sbourdeauducq, 2015-06-26:
There is ``artiq.coredevice.dds.DDS.init()``, but as Robert points out, you probably should not really need to use it.

jboulder, 2015-06-26:
At present the reset button on the Kintex7 eval board is accessible with
the FMC connector attached. This may not be the case in the future if
things get boxed up.

At present the reset button on the Pipistrello is difficult to access when
the TTL/DDS bus adapter PCB is attached to its headers.

Its good to know about artiq.coredevice.dds.DDS.init(). -Joe

jordens, 2015-06-26:
This is about DDS reset/init. FPGA reset/init is a different problem. But if somebody needs it it is always a bug and never a solution.

sbourdeauducq, 2015-07-05:
I believe this can be resolved by using ``artiq.coredevice.dds.DDS.init()`` in the worst case. Please open a new issue if you have trouble resetting the FPGA.

closed 2015-07-05 by sbourdeauducq