m-labs/artiq #36: question: should we compensate for latency of FUD to DAC output of DDS (pipeline delay)?

jboulder, 2015-06-20:


jboulder, 2015-06-20:
We should do this. The AD9858 specification says this is 83 cycles of SYS_CLK (1 GHz). This is a fixed delay and can be trivially compensated for by Artiq. It's one less thing a user has to worry about.

jordens, 2015-06-21:
Rationale?
Doing that in gateware does not gain anything. My idea was always to define it in the `ddb` and then handle it in software. But that needs spec'ing.

closed 2015-06-21 by jordens

jboulder, 2015-06-21:
A straightforward way of achieving pipeline delay compensation is using the gateware. If you prefer to implement this in software please do so. I'd like to be able to report at the group meeting on Monday that the AD9858 implementation in Artiq passes the test suite. Without pipeline delay compensation it does not.

jordens, 2015-06-21:
Could you point me to a test in the testsuite that needs this to pass? It seems naive to compensate w.r.t. DAC output as the reference plane if there are (frequency dependent even!) delays in the trafos, amplifiers, cables after the DAC output.