m-labs/artiq #38: DDS test mode... 1111

jboulder, 2015-06-21:
Using KC705 in Penning lab. Connected to legacy Penning hardware using FMC-LVDS adapter. 

In test mode run DDS tests on AD9858. Generates all 1111's.

sbourdeauducq, 2015-06-21:
Happened whenever a DDS register was read - always returned 0xff. Might be a hardware problem, and this did not happen in previous tests.

jboulder, 2015-06-25:
Repeated pattern is not noise. Using Penning lab test setup.
```bash
test> clksrc 1
test> ddstest 5
readback fail on DDS 0, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 0, 0xffffffff != 0x55555555
readback fail on DDS 0, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 0, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 0, 0xffffffff != 0x00000000
readback fail on DDS 0, 0xffffffff != 0x12345678
readback fail on DDS 0, 0xffffffff != 0x87654321
readback fail on DDS 0, 0xffffffff != 0x0000ffff
readback fail on DDS 0, 0xffffffff != 0xffff0000
readback fail on DDS 0, 0xffffffff != 0x00ff00ff
readback fail on DDS 0, 0xffffffff != 0xff00ff00
readback fail on DDS 1, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 1, 0xffffffff != 0x55555555
readback fail on DDS 1, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 1, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 1, 0xffffffff != 0x00000000
readback fail on DDS 1, 0xffffffff != 0x12345678
readback fail on DDS 1, 0xffffffff != 0x87654321
readback fail on DDS 1, 0xffffffff != 0x0000ffff
readback fail on DDS 1, 0xffffffff != 0xffff0000
readback fail on DDS 1, 0xffffffff != 0x00ff00ff
readback fail on DDS 1, 0xffffffff != 0xff00ff00
readback fail on DDS 2, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 2, 0xffffffff != 0x55555555
readback fail on DDS 2, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 2, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 2, 0xffffffff != 0x00000000
readback fail on DDS 2, 0xffffffff != 0x12345678
readback fail on DDS 2, 0xffffffff != 0x87654321
readback fail on DDS 2, 0xffffffff != 0x0000ffff
readback fail on DDS 2, 0xffffffff != 0xffff0000
readback fail on DDS 2, 0xffffffff != 0x00ff00ff
readback fail on DDS 2, 0xffffffff != 0xff00ff00
readback fail on DDS 3, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 3, 0xffffffff != 0x55555555
readback fail on DDS 3, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 3, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 3, 0xffffffff != 0x00000000
readback fail on DDS 3, 0xffffffff != 0x12345678
readback fail on DDS 3, 0xffffffff != 0x87654321
readback fail on DDS 3, 0xffffffff != 0x0000ffff
readback fail on DDS 3, 0xffffffff != 0xffff0000
readback fail on DDS 3, 0xffffffff != 0x00ff00ff
readback fail on DDS 3, 0xffffffff != 0xff00ff00
readback fail on DDS 4, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 4, 0xffffffff != 0x55555555
readback fail on DDS 4, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 4, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 4, 0xffffffff != 0x00000000
readback fail on DDS 4, 0xffffffff != 0x12345678
readback fail on DDS 4, 0xffffffff != 0x87654321
readback fail on DDS 4, 0xffffffff != 0x0000ffff
readback fail on DDS 4, 0xffffffff != 0xffff0000
readback fail on DDS 4, 0xffffffff != 0x00ff00ff
readback fail on DDS 4, 0xffffffff != 0xff00ff00
readback fail on DDS 5, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 5, 0xffffffff != 0x55555555
readback fail on DDS 5, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 5, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 5, 0xffffffff != 0x00000000
readback fail on DDS 5, 0xffffffff != 0x12345678
readback fail on DDS 5, 0xffffffff != 0x87654321
readback fail on DDS 5, 0xffffffff != 0x0000ffff
readback fail on DDS 5, 0xffffffff != 0xffff0000
readback fail on DDS 5, 0xffffffff != 0x00ff00ff
readback fail on DDS 5, 0xffffffff != 0xff00ff00
[... 349 lines cut ...]
readback fail on DDS 5, 0xffffffff != 0xffff0000
readback fail on DDS 5, 0xffffffff != 0x00ff00ff
readback fail on DDS 5, 0xffffffff != 0xff00ff00
readback fail on DDS 6, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 6, 0xffffffff != 0x55555555
readback fail on DDS 6, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 6, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 6, 0xffffffff != 0x00000000
readback fail on DDS 6, 0xffffffff != 0x12345678
readback fail on DDS 6, 0xffffffff != 0x87654321
readback fail on DDS 6, 0xffffffff != 0x0000ffff
readback fail on DDS 6, 0xffffffff != 0xffff0000
readback fail on DDS 6, 0xffffffff != 0x00ff00ff
readback fail on DDS 6, 0xffffffff != 0xff00ff00
readback fail on DDS 7, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 7, 0xffffffff != 0x55555555
readback fail on DDS 7, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 7, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 7, 0xffffffff != 0x00000000
readback fail on DDS 7, 0xffffffff != 0x12345678
readback fail on DDS 7, 0xffffffff != 0x87654321
readback fail on DDS 7, 0xffffffff != 0x0000ffff
readback fail on DDS 7, 0xffffffff != 0xffff0000
readback fail on DDS 7, 0xffffffff != 0x00ff00ff
readback fail on DDS 7, 0xffffffff != 0xff00ff00

sbourdeauducq, 2015-06-26:
Can you try with a single read, and check with a logic analyzer if 1) the FPGA provides a valid read pulse on rd_n 2) following the pulse, the DDS system provides valid levels on the data pins after going through the SCSI cables, adapter, etc.

sbourdeauducq, 2015-08-21:
This "1111...." problem is not present on the AD9914, which uses very similar code there. Are you sure this is not just a problem with the notoriously unreliable adapter/cables? Is there a way you could connect a logic analyzer on this thing? Otherwise, maybe we should integrate miscope and enable it by default on the DDS pins... DDS debugging is a lot of tedious guesswork otherwise.

sbourdeauducq, 2015-08-22:
Just passed 100K cycles of ddstest on the AD9914...

jboulder, 2015-09-01:
Still not working right. 

```bash
rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                     np_1520  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
llvmlite-artiq            0.5.1                    py34_4  

rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq$ flterm --port /dev/ttyUSB1[FLTERM] Starting...
�
MiSoC BIOS   http://m-labs.hk
(c) Copyright 2007-2014 Sebastien Bourdeauducq
Revision a4808ace built Sep  1 2015 12:41:48

BIOS CRC passed (6fe8524e)
Running on MiSoC rev. a4808ace (sysid:K7) at 125MHz
Initializing SDRAM...
Write leveling: 13  14* 12  11   8   7   5   6  completed
Read bitslip: 7 6 5 4 3 
Read delays: 7:01-10  6:01-11  5:03-12  4:03-13  3:10-21  2:00-06  1:00-09  0:00-10  completed
Memtest OK
Automatic boot in 2 seconds...
Q/ESC: abort boot
F7:    boot from serial
F8:    boot from network
Booting from flash...
Loading 150640 bytes from flash...
Executing booted program.
ARTIQ runtime built Sep  1 2015 12:42:13

Startup RTIO clock: internal
Press 't' to enter test mode...
Entering test mode.
test> 
test> help
Available commands:
help            - this message
clksrc <n>      - select RTIO clock source
ttloe <n> <v>   - set TTL output enable
ttlo <n> <v>    - set TTL output value
ddssel <n>      - select a DDS
ddsinit         - reset, config, FUD DDS
ddsreset        - reset DDS
ddsw <a> <d>    - write to DDS register
ddsr <a>        - read DDS register
ddsfud          - pulse FUD
ddsftw <n> <d>  - write FTW
ddstest <n> <c> - perform test sequence on DDS
leds <n>        - set LEDs
fserase         - erase flash storage
fswrite <k> <v> - write to flash storage
fsread <k>      - read flash storage
fsremove <k>    - remove a key-value record from flash storage
fstest          - run flash storage tests. WARNING: erases the storage area
test> ddsreset
test> ddsinit
test> ddstest 1 all
readback fail on DDS 0, 0xffffffff != 0xaaaaaaaa
readback fail on DDS 0, 0xffffffff != 0x55555555
readback fail on DDS 0, 0xffffffff != 0xa5a5a5a5
readback fail on DDS 0, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 0, 0xffffffff != 0x00000000
readback fail on DDS 0, 0xffffffff != 0x12345678
readback fail on DDS 0, 0xffffffff != 0x87654321
readback fail on DDS 0, 0xffffffff != 0x0000ffff
readback fail on DDS 0, 0xffffffff != 0xffff0000
readback fail on DDS 0, 0xffffffff != 0x00ff00ff
[... 56 lines cut ...]
readback fail on DDS 6, 0xfefefefe != 0xaaaaaaaa
readback fail on DDS 6, 0xfefefefe != 0x55555555
readback fail on DDS 6, 0xfdfdfdfd != 0xa5a5a5a5
readback fail on DDS 6, 0xffffffff != 0x5a5a5a5a
readback fail on DDS 6, 0xfefefefe != 0x00000000
readback fail on DDS 6, 0xfffcffff != 0x12345678
readback fail on DDS 6, 0xbffdfbf9 != 0x87654321
readback fail on DDS 6, 0xfeffffff != 0x0000ffff
readback fail on DDS 6, 0xfffffefe != 0xffff0000
readback fail on DDS 6, 0xffffffff != 0x00ff00ff
readback fail on DDS 6, 0xffffffff != 0xff00ff00
readback fail on DDS 7, 0xfefffefe != 0xaaaaaaaa
readback fail on DDS 7, 0xfffffeff != 0x55555555
readback fail on DDS 7, 0xfefefefe != 0xa5a5a5a5
readback fail on DDS 7, 0xfefafefe != 0x5a5a5a5a
readback fail on DDS 7, 0xfefefefe != 0x00000000
readback fail on DDS 7, 0xfafcfffe != 0x12345678
readback fail on DDS 7, 0xff7d7bfd != 0x87654321
readback fail on DDS 7, 0xffffffff != 0x0000ffff
readback fail on DDS 7, 0xfffffeff != 0xffff0000
readback fail on DDS 7, 0xffffffff != 0x00ff00ff
readback fail on DDS 7, 0xfffefffe != 0xff00ff00
test> 

```

jboulder, 2016-03-20:
I no longer have access to this hardware to conduct further tests.  @kgilmo is the main ARTIQ guy on that hardware. My current expectation is that these errors are a result of the buggy LVDS bus used in this legacy system and not an ARITQ-specific issue. If @kgilmo doesn't want to take this on the issue should be closed.

closed 2016-03-29 by sbourdeauducq