m-labs/artiq #39: Make Qt + asyncio + UDP work on Windows

sbourdeauducq, 2015-06-21:
The GUI needs UDP to talk to the core device in the TTL and DDS panel. On Windows, the quamash event loop is based on ProactorEventLoop which does not support UDP.

jordens, 2015-08-22:
I had a look at how this could be done a while ago. Doing this the "right" way, that is within tulip/asyncio is close to impossible. The common denominator event loop for qt/quamash/tulip is the Proactor/IOCP stuff.

That said: The "inj" part should be work-around-able. The name lookup can be done asyncio-style and then sending udp through a regular socket is non-blocking. The "mon" part is a bit harder. Delegating moninj to a slave process/thread that translates udp to stdout/stdin might be ok.

sbourdeauducq, 2015-08-22:
Did you have a look at how exactly Microsoft (or whoever is responsible) messed up proactor/IOCP enough for UDP to be that difficult?

jordens, 2015-08-22:
IOCP-UDP seems to have zero users in general. The best pointers that I have are that one would have to marry something like this: http://members.gamedev.net/edxLabs/network/WHSPSUSUIOCP.zip with this: https://github.com/python/asyncio/blob/master/overlapped.c

sbourdeauducq, 2015-08-24:
84b0760 disables moninj on windows so the GUI should at least start.

sbourdeauducq, 2015-10-13:
If we use UDP for logging, we're going to need that as well in the master and the controller manager. They also need to use ProactorEventLoop due to their use of subprocesses.

sbourdeauducq, 2015-10-14:
There is an existing issue in the Python tracker:
https://bugs.python.org/issue23295

sbourdeauducq, 2016-01-26:
@jordens Who wants this in 1.0?

sbourdeauducq, 2016-01-26:
This will require a Python patch, which from past experience with multibind takes 10 months to merge.

whitequark, 2016-01-26:
@sbourdeauducq You are OK with shipping modified Qt, but not modified Python? Python is way easier to build and distribute.

sbourdeauducq, 2016-01-26:
I don't like either, but the Qt bugs consistently produced ugly display artifacts whereas this Python issue only reduces the features available on Windows. The Qt bugfixes should go directly upstream, we have the workarounds in place in the meantime.

jordens, 2016-01-26:
@sbourdeauducq I put this into the 1.0 milestone because this is severely limiting the usability of the GUI on windows. The latter would be something we want for 1.0.

sbourdeauducq, 2016-04-11:
Done with threads. Slightly hacky, but we should end up rewriting this with TCP at some point anyway.

closed 2016-04-11 by sbourdeauducq