m-labs/artiq #44: Use PPP to connect to core device in serial mode

sbourdeauducq, 2015-06-21:
Compared to the current solution, PPP adds robustness, error detection/correction and support for multiplexing monitoring/injection data, with maximum code reuse.

PPP should already be supported by lwip.

(This is low priority)

jordens, 2015-06-21:
How sad:

> Serial Line Internet Protocol (SLIP) option is not available in Windows 7.
> However, if you have Windows 7 Professional or Ultimate, you may try using Windows XP mode for SLIP connection.

[Microsoft](http://answers.microsoft.com/en-us/windows/forum/windows_7-networking/dial-up-slip-connection-in-windows-7/24d3913a-46f1-4673-a358-1e80366675df)

sbourdeauducq, 2015-06-21:
PPP should be fine, no? AOHell even still make millions from dial-up connections in the US.

jordens, 2015-06-21:
PPP is fine. Just reminiscing.

sbourdeauducq, 2015-06-27:
9492802 makes the PPP connection work as well as most of ICMP and UDP. TCP is unusably slow, and the wireshark capture shows excessive retransmissions. ``ping -s 1000 -f 10.0.0.2`` shows excessive packet losses.

On the PC side, use this command:
``pppd /dev/ttyUSB1 115200 noauth  nodetach local nocrtscts 10.0.0.1:10.0.0.2``

sbourdeauducq, 2015-06-27:
Could be a flow control issue at the serial port level.

sbourdeauducq, 2015-06-27:
Increasing the UART software FIFO buffer sizes in MiSoC to 8192 reduces the ``ping -s 1000 -f 10.0.0.2`` packet loss rate and does nothing for TCP.

jordens, 2015-06-29:
Nice!
Is the TCP problem dependent of congestion or does it also happen at light load?
Does `pppd` ignore stuff sufficiently and resynchronize itself if left running through a core device boot or does it need to be started after runtime boot?

sbourdeauducq, 2015-06-29:
Also happens at light load.

felixheld, 2015-11-24:
i tracked down the problem: the application doesn't read the software receive buffer fast enough and when the two rx fifo pointers point to the same location (== the buffer is full) it stops reading bytes from the hardware buffer in the interrupt handlers for a long enough time that some data gets lost.
i'd add a threshold at which a xoff byte is sent, so that the ftdi stops sending data in time and no data gets lost.

jordens, 2015-11-25:
I had looked into this (ppp, the serial buffering, top half and bottom half, the various ringbuffer corner cases) a while ago. The two rx ringbuffer pointers equal means fifo empty. Not full.

And since the errors also happen for rather huge buffers (i tried 8k IIRC), the stack must be horribly slow to handle packets at only such a slow rate (~ 100 packets/s if my numbers are right).

felixheld, 2015-11-25:
in https://github.com/m-labs/misoc/blob/master/misoc/software/libbase/uart.c#L34 it is checked if the *next* byte writing address is equal the current read address. So those two addresses being equal means the software buffer is full; if new data was written to the buffer, old data would be overwritten.
The buffer would be empty, if the current write address and the current read address are equal, but the the next write address wouldn't be equal to the current read address.
i do agree though that the stack is ways slower than i'd expect it to be. so it is somehow likely that this bug is just the effect of some other bug. problem is that there is no proper debugging support...

sbourdeauducq, 2015-11-25:
What would be "proper debugging support" and how exactly would it help in this case?

felixheld, 2015-11-25:
i was thinking about instruction trace support to identify where the code consumes that much cycles that it isn't able to process those maybe 100 packets per second and breakpoint support to have a closer look inside those functions

sbourdeauducq, 2015-11-26:
Use the timer and a second UART? FWIW there was a similar issue with Ethernet that was fixed in 110f7bce646e24453b291741dd7bfb68837592fa. Of course, you'd think lwip would print error messages in those cases...

sbourdeauducq, 2015-11-26:
8d1663394be0e602ac7ebb6fcd62e69d4982cf5f was also a lot of fun. Rounding issue in the lwipopts example, and no error printed.

jordens, 2015-11-26:
If anyone wants it for this, I have a half working UART-over-BSCAN-over-openocd around.

jordens, 2016-01-19:
Much better now.

```
(artiq)rjordens@688rj1:~/work/nist/artiq (ppp)$ sudo ping -f 10.0.0.2
PING 10.0.0.2 (10.0.0.2) 56(84) bytes of data.
.^C
--- 10.0.0.2 ping statistics ---
2121 packets transmitted, 2120 received, 0% packet loss, time 29678ms
rtt min/avg/max/mdev = 15.646/16.088/31.971/0.713 ms, pipe 3, ipg/ewma 13.999/15.914 ms
```

But it looks like there is a different buffering issue now (this is traffic due to an `artiq_run`):

```
Capturing on 'ppp0'
  1   0.000000     10.0.0.1 -> 10.0.0.2     TCP 76 36266 > 1381 [SYN, ECN, CWR] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=764278566 TSecr=0 WS=128
  2   0.010347     10.0.0.2 -> 10.0.0.1     TCP 60 1381 > 36266 [SYN, ACK] Seq=0 Ack=1 Win=16384 Len=0 MSS=1460
  3   0.010399     10.0.0.1 -> 10.0.0.2     TCP 56 36266 > 1381 [ACK] Seq=1 Ack=1 Win=29200 Len=0
  4   0.010573     10.0.0.1 -> 10.0.0.2     TCP 70 36266 > 1381 [PSH, ACK] Seq=1 Ack=1 Win=29200 Len=14
  5   0.025314     10.0.0.2 -> 10.0.0.1     TCP 56 1381 > 36266 [ACK] Seq=1 Ack=15 Win=16370 Len=0
  6   0.025348     10.0.0.1 -> 10.0.0.2     TCP 65 36266 > 1381 [PSH, ACK] Seq=15 Ack=1 Win=29200 Len=9
  7   0.034319     10.0.0.2 -> 10.0.0.1     TCP 85 1381 > 36266 [PSH, ACK] Seq=1 Ack=24 Win=16361 Len=29
  8   0.034435     10.0.0.1 -> 10.0.0.2     TCP 56 36266 > 1381 [ACK] Seq=24 Ack=30 Win=29200 Len=0
  9   0.034840     10.0.0.1 -> 10.0.0.2     TCP 65 36266 > 1381 [PSH, ACK] Seq=24 Ack=30 Win=29200 Len=9
 10   0.045327     10.0.0.2 -> 10.0.0.1     TCP 56 [TCP Previous segment not captured] 1381 > 36266 [ACK] Seq=31 Ack=33 Win=16352 Len=0
 11   0.045365     10.0.0.1 -> 10.0.0.2     TCP 57 36266 > 1381 [PSH, ACK] Seq=33 Ack=30 Win=29200 Len=1
11  12   0.254222     10.0.0.1 -> 10.0.0.2     TCP 57 [TCP Keep-Alive] 36266 > 1381 [PSH, ACK] Seq=33 Ack=30 Win=29200 Len=1
 13   0.260343     10.0.0.2 -> 10.0.0.1     TCP 56 1381 > 36266 [ACK] Seq=31 Ack=34 Win=16351 Len=0
13  14   0.748318     10.0.0.2 -> 10.0.0.1     TCP 85 [TCP Retransmission] 1381 > 36266 [SYN, RST, ACK] Seq=1 Ack=34 Win=16351 Len=29
 15   0.770317     10.0.0.2 -> 10.0.0.1     TCP 56 [TCP Keep-Alive] 1381 > 36266 [ACK] Seq=30 Ack=34 Win=16351 Len=0
15  16   1.170289     10.0.0.2 -> 10.0.0.1     TCP 56 [TCP Keep-Alive] 1381 > 36266 [ACK] Seq=30 Ack=34 Win=16351 Len=0
 17   1.570360     10.0.0.2 -> 10.0.0.1     TCP 56 [TCP Keep-Alive] 1381 > 36266 [ACK] Seq=30 Ack=34 Win=16351 Len=0
17  18   1.970337     10.0.0.2 -> 10.0.0.1     TCP 56 1381 > 36266 [RST, ACK] Seq=31 Ack=34 Win=16384 Len=0
 19   1.970387     10.0.0.1 -> 10.0.0.2     TCP 56 [TCP Dup ACK 12#1] 36266 > 1381 [ACK] Seq=34 Ack=30 Win=29200 Len=0
 20   1.979368     10.0.0.2 -> 10.0.0.1     TCP 56 1381 > 36266 [RST, ACK] Seq=30 Ack=34 Win=16384 Len=0
20 ^C
```

jordens, 2016-01-19:
And it would be cool to use Florent's parallel FIFO core instead of the uart. https://github.com/enjoy-digital/liteusb/blob/master/liteusb/phy/ft245.py

sbourdeauducq, 2016-01-19:
The main point of this PPP support is portability, not performance. In that case, a common and simple 2-pin serial interface wins.

jordens, 2016-01-19:
It's exchangeable and using the parallel interface does not affect portability. The Opal Kelly boards are byte-wise as well and would need this abstraction anyway.

jordens, 2016-01-25:
Closed with 2c847b3. Remaining items:

* VJ compression should be fixed in lwip at some point and the recommended in the manual.
* The parallel interface for the synchronous ft2232h fifo and the cypress chip for the Opal Kelly boards would still be nice.

closed 2016-01-25 by jordens

jordens, 2016-02-25:
If the uart speed is cranked up to 921600 Baud, the uart rx ringbuffer and
the ppp/tcp stack can't keep up. Lots of lost segments, retransmissions and/or duplicate acks. Increasing the rx ringbufffer to 2048 helps a lot (but the uart is not easily reconfigurable and
the code as it is written has to fit the sram of the bios as well).