m-labs/artiq #46: AD9858  PHASE_MODE_CONTINUOUS broken

jboulder, 2015-06-22:
On 6/19 PHASE_MODE_CONTINUOUS was working but PHASE_MODE_TRACKING was broken.
On 6/20 PHASE_MODE_TRACKING was fixed but PHASE_MODE_CONTINUOUS was not retested.
On 6/21 PHASE_MODE_TRACKING works but PHASE_MODE_CONTINUOUS is broken.

[image]

```python
class TestDDSPhase(Experiment, AutoDB):
	class DBKeys:
		core = Device()
		dds_bus = Device()
		dds0 = Device()
		dds1 = Device()
		ttl0 = Device()
		ttl1 = Device()

	@kernel
	def run(self):
		self.dds0.set_phase_mode(PHASE_MODE_CONTINUOUS)
		self.dds1.set_phase_mode(PHASE_MODE_CONTINUOUS)
		for i in range(10):
			# force DDSs to be phase synchronized at outset of experiment
			# this is a temporary override of the default phase mode
			with self.dds_bus.batch:
				self.dds0.set(3*MHz, phase_mode=PHASE_MODE_ABSOLUTE, phase_offset=0)
				self.dds1.set(3*MHz, phase_mode=PHASE_MODE_ABSOLUTE, phase_offset=0)

			# emit experiment start pulse on TTL0
			self.ttl0.pulse(1*us)

			# demonstrate simple spin-echo type pulse sequence
			self.dds0.set(3*MHz, phase_offset=0, phase_mode=PHASE_MODE_CONTINUOUS)
			self.ttl1.pulse(1*us) # first pi/2 pulse
			self.dds0.set(3*MHz, phase_offset=0.25, phase_mode=PHASE_MODE_CONTINUOUS)
			delay(2*us) # arm time
			self.ttl1.pulse(2*us) # middle pi pulse
			self.dds0.set(3*MHz, phase_offset=0, phase_mode=PHASE_MODE_CONTINUOUS)
			delay(2*us) # arm time
			self.ttl1.pulse(1*us) # final pi/2 pulse

			# delay to let Core_uP catch up
			delay(1*ms)

jboulder, 2015-06-22:
Here's a clearer illustration. 

[image]

```python
class TestDDSPhase(Experiment, AutoDB):
	class DBKeys:
		core = Device()
		dds_bus = Device()
		dds0 = Device()
		dds1 = Device()
		ttl0 = Device()
		ttl1 = Device()

	@kernel
	def run(self):
		self.dds0.set_phase_mode(PHASE_MODE_CONTINUOUS)
		self.dds1.set_phase_mode(PHASE_MODE_CONTINUOUS)
		for i in range(10):
			# force DDSs to be phase synchronized at outset of experiment
			# this is a temporary override of the default phase mode
			with self.dds_bus.batch:
				self.dds0.set(3*MHz, phase_mode=PHASE_MODE_ABSOLUTE, phase_offset=0)
				self.dds1.set(3*MHz, phase_mode=PHASE_MODE_ABSOLUTE, phase_offset=0)

			# emit experiment start pulse on TTL0
			self.ttl0.pulse(1*us)

			# demonstrate simple spin-echo type pulse sequence
			# self.dds0.set(3*MHz, phase_offset=0, phase_mode=PHASE_MODE_CONTINUOUS)
			self.ttl1.pulse(.5*us) # first pi/2 pulse
			self.dds0.set(3*MHz, phase_offset=0.25, phase_mode=PHASE_MODE_CONTINUOUS)
			
			delay(1*us) # arm time
			
			self.ttl1.pulse(1*us) # middle pi pulse
			self.dds0.set(3*MHz, phase_offset=0, phase_mode=PHASE_MODE_CONTINUOUS)
			
			delay(1*us) # arm time
			
			self.ttl1.pulse(0.5*us) # final pi/2 pulse

			# delay to let Core_uP catch up
			delay(1*ms)

jboulder, 2015-06-22:
Quick test to see if this behavior is related to the timing of the FTW. 
1) reset power on DDS box and reset DDS
2) run program above
3) look on scope

[image]
[image]
[image]
[image]

sbourdeauducq, 2015-06-22:
Do you still have any issue with a board flashed with the latest runtime from the conda packages? I pushed the bugfixes only recently. Do you have this commit: https://github.com/m-labs/artiq/commit/87ea1433d3bef86ad5b3a06651b6d4d880cdc343

jboulder, 2015-06-23:
> this commit: 87ea143

I will try to two suggestions you made. It will be a day or two. We are
back running on the old DDS system at the moment. -Joe

jboulder, 2015-06-24:
OK. Resolved in today's conda build. 

[image]

sbourdeauducq, 2015-06-24:
Good. Thanks for the quick re-test!

closed 2015-06-24 by sbourdeauducq