m-labs/artiq #48: needs compiler optimization

jboulder, 2015-06-22:
This experiment script was marked for bug reporting. SB, do you recall the problem? I'm applying a 1 MHz external pulse train to PMT1. 

timestamps_compiler_opt_fail.py

```python
from artiq import *

class LED(Experiment, AutoDB):
	class DBKeys:
		core = Device()
		ttl0 = Device()
		pmt1 = Device() 

	def process(self, x):
		print(x)
		print([t1-t0 for t0, t1 in zip(x, x[1:])])

	@kernel
	def run(self):
		L = 100
		buf = [int64(0) for _ in range(L)]
		self.pmt1.gate_rising(1000*ms)
		count = 0
		for i in range(L):
			t = time_to_cycles(self.pmt1.timestamp())
			if t > 0:
				count += 1
			buf[i] = t 
		print(count)
		self.process(buf)
```
Output is below.
```python
rabi2@vboxartiq:~/artiq/examples/master$ artiq_run repository/timestamps_compiler_opt_fail.py 
100
[244726, 369726, 494726, 619726, 744726, 869726, 994726, 1119726, 1244726, 1369726, 1494726, 1619726, 1744726, 1869726, 1994726, 2119726, 2244726, 2369726, 2494726, 2619726, 2744726, 2869726, 2994726, 3119726, 3244726, 3369726, 3494726, 3619726, 3744726, 3869725, 3994725, 4119725, 4244725, 4369725, 4494725, 4619725, 4744725, 4869725, 4994725, 5119725, 5244725, 5369725, 5494725, 5619725, 5744725, 5869725, 5994725, 6119725, 6244725, 6369725, 6494725, 6619725, 6744725, 6869725, 6994725, 7119725, 7244725, 7369725, 7494725, 7619725, 7744725, 7869725, 7994725, 8119724, 8244724, 8369724, 8494724, 8619724, 8744724, 8869724, 8994724, 9119724, 9244724, 9369724, 9494724, 9619724, 9744724, 9869724, 9994724, 10119724, 10244724, 10369724, 10494724, 10619724, 10744724, 10869724, 10994724, 11119724, 11244724, 11369724, 11494724, 11619724, 11744724, 11869724, 11994724, 12119724, 12244723, 12369723, 12494723, 12619723]
[125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 124999, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 124999, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 125000, 124999, 125000, 125000, 125000]

sbourdeauducq, 2015-07-13:
As I recall, that broke with a large L, similarly to issue #49. Is that still happening?

whitequark, 2015-12-17:
I've updated the code to use modern ARTIQ:
```python
from artiq import *

class LED(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")
        self.setattr_device("pmt1")

    def process(self, buf):
        print(buf)

    @kernel
    def run(self):
        L = 20000
        buf = [int(0, width=64) for _ in range(L)]
        self.pmt1.gate_rising(1000*ms)
        count = 0
        for i in range(20000):
            t = self.pmt1.timestamp_mu()
            if t > 0:
                count += 1
            buf[i] = t
        print(count)
        self.process(buf)
```

It successfully outputs a long sequence of [-1]s with L<=20000, and raises an RTIOUnderflow with Ls>=30000. Is that the expected behavior?

sbourdeauducq, 2015-12-18:
No, -1 means no rising edges were detected - did you connect the 1MHz generator?

whitequark, 2015-12-18:
No, I have no idea what to connect, and also I'm not in the lab...

whitequark, 2015-12-18:
I tried to add a TTLClockGen on 19th channel, still no effect:
```python
from artiq import *

class LED(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")
        self.setattr_device("ttl_sma")
        self.setattr_device("ttl_clock")

    def process(self, buf):
        print(buf)

    @kernel
    def run(self):
        L = 20000
        buf = [int(0, width=64) for _ in range(L)]
        self.ttl_sma.gate_rising(1000*ms)
        self.ttl_clock.set_mu(1000)
        count = 0
        for i in range(20000):
            t = self.ttl_sma.timestamp_mu()
            if t > 0:
                count += 1
            buf[i] = t
        print(count)
        self.process(buf)
```

sbourdeauducq, 2015-12-19:
The 10MHz clock on SMA I'm talking about is built into the gateware and driven here:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/kc705.py#L37

The SMA TTL is here:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/kc705.py#L168

The clock generator on channel 19 is a different pin (it's on the FMC and relatively hard to connect), and it won't feed into the SMA TTL unless you connect it physically.

whitequark, 2016-01-05:
@sbourdeauducq, so would you test this?

sbourdeauducq, 2016-01-15:
Affected by a similar problem as #228 now...

whitequark, 2016-02-15:
#228 was fixed, does this work now?

sbourdeauducq, 2016-02-15:
I'm pretty sure this works, but I'd be paranoid and test it anyway.

sbourdeauducq, 2016-02-22:
Works but timestamps are affected by #286.

closed 2016-02-22 by sbourdeauducq