m-labs/artiq #49: more_compiler_bugs.py

jboulder, 2015-06-22:
This was marked for compiler optimization. 

```python

from artiq import *

class LED(Experiment, AutoDB):
	class DBKeys:
		core = Device()

	@kernel
	def run(self):
		L = 10000
		buf = [int64(0) for _ in range(L)]
		count = 0
		for i in range(L):
			if buf[i] > 0:
				count += 1
		print(count)

sbourdeauducq, 2015-07-13:
Works totally fine here on Pipistrello. Here is the code converted to the new device API:

```python
from artiq import *

class Test(EnvExperiment):
    def build(self):
        self.attr_device("core")

    @kernel
    def run(self):
        L = 10000
        buf = [int64(0) for _ in range(L)]
        count = 0
        for i in range(L):
            if buf[i] > 0:
                count += 1
        print(count)
```

Do you still have problems with it?

sbourdeauducq, 2015-07-13:
As I remember, this was not a compiler optimization problem, but an instability problem. Might have been solved with the concurrent memory access correction.

sbourdeauducq, 2015-07-25:
Just tested OK on KC705 as well. Reopen if you still have problems.

closed 2015-07-25 by sbourdeauducq