m-labs/artiq #50: rename llvmlite-or1k package name from llvmlite to llvmlite_or1k

jordens, 2015-06-22:
14:42 < rjo> ysionneau: good that you added the -or1k suffix to the llvmlite package.
14:43 < rjo> ysionneau: but: it would be really great if we could also install that under that name
             (llvmlite_or1k) in the package tree
14:43 < rjo> because then you can actually install your regular native llvmlite in parallel with the
             one linked against the or1k-llvm
14:45 < rjo> afaict there would just be a bunch of s/llvmlite/llvmlite_or1k/ in the artiq source.
14:45 < rjo> the native llvmlite is used by numba, which i tend to use.
14:46 < ysionneau> ah ok
14:46 < rjo> ok if i file an issue for this to keep track?
14:46 < ysionneau> yes sure
14:46 < ysionneau> but I'm not sure I understand exactly how to do that
14:46 < ysionneau> in $HOME/miniconda3/pkgs the name of the package is already -or1k
14:47 < rjo> yes. but in the install tree.
14:47 < ysionneau> ah in pkgs/llvmlite-or1k-0.2.1-py34_0/lib/python3.4/site-packages
14:47 < ysionneau> ok
14:47 < ysionneau> I guess it's a setup.py option?
14:47 < ysionneau> to change the package name
14:47 < rjo> less an option. more a field.
14:47 < rjo> exactly.
14:48 < rjo> would need to patch that.

fallen, 2015-07-12:
I was able to generate a conda package for linux-32 and linux-64 (and run the transforms.py unit test).
But it won't compile under win32 (with Visual Studio) if I set -DBUILD_SHARED_LIBS=ON :/
@whitequark any idea how to fix the win32 build when generating shared libs?

whitequark, 2015-07-12:
What's the error message?

fallen, 2015-07-12:
Here is the entire compile log on win-32: http://pastebin.com/A94KMNVG

fallen, 2015-07-12:
some words are in French, sorry about that.

whitequark, 2015-07-12:
ok, let me build it locally

fallen, 2015-07-12:
If you want to reproduce in the same conditions, from artiq root you can do : 
cd conda
conda build llvmdev-or1k
(you need either anaconda or miniconda + visual studio installed)

whitequark, 2015-07-12:
ugh, it requires win8.1. don't have it...

fallen, 2015-07-12:
What is requiring  win8.1 ? I only have win 7

whitequark, 2015-07-12:
VS2013: https://github.com/m-labs/artiq/blob/master/conda/llvmdev-or1k/bld.bat#L6

whitequark, 2015-07-12:
I don't have win7 either, anyway

fallen, 2015-07-12:
I have VS2013 in my win 7 vm.

whitequark, 2015-07-12:
hm, the MS website and the installer insist it requires 8.1

whitequark, 2015-07-12:
do you want the x86 or amd64 one?

fallen, 2015-07-12:
For now I'm building only win32 packages so x86 one.

whitequark, 2015-07-12:
ok. fyi win32 is the ABI name for the x86_64 system as well, so that's a bit confusing

fallen, 2015-07-12:
Ah ok, didn't know that, that's a bit misleading.

whitequark, 2015-07-12:
By the way, you really should build Debug binaries, or at least MinSizeRel if a few GB of debug info is too much. The optimizations are taking a lot of time and there's little use to them.

whitequark, 2015-07-12:
Reproduced.

whitequark, 2015-07-12:
Haha, so, well. "Shared libraries are not supported on Windows" http://llvm.org/docs/CMake.html

fallen, 2015-07-13:
Ooook, awesome. Thanks ^^

fallen, 2015-07-13:
Would you recommend going back to using static library for linux+win? Or just for Windows?

whitequark, 2015-07-13:
I would think that the advantage of consistency across platforms outweighs any gains from using shared libraries.

jordens, 2015-07-13:
I would suspect that the numba folks have put a lot of thought into these questions. Maybe just sticking what they do is smart.

whitequark, 2015-07-13:
They just use llvmlite.

jordens, 2015-07-13:
They package llvm and llvmlite for all OS.

fallen, 2015-07-13:
Fixed in https://github.com/m-labs/artiq/commit/90ba9f7bbf02537103b23214d23fc12577049628

sbourdeauducq, 2015-07-14:
The patch that renames the package should not be called "add all targets".

fallen, 2015-07-15:
@jboulder could you run the anaconda python interpreter and run : import llvmlite_or1k.binding as llvm please?
If it does not work, it means your llvmlite-or1k conda package is not up to date.
In that case I would suggest you do something like:
conda remove llvmlite-or1k
rm -rf $HOME/anaconda3/pkgs/llvmlite-or1k-0.2.1*
conda install llvmlite-or1k

jboulder, 2015-07-15:
```bash
$ which ipython
/home/rabi2/anaconda3/bin/ipython
$ ipython
```python
In [1]:  import llvmlite_or1k.binding as llvm
ImportError: No module named 'llvmlite_or1k'
```
Follow suggested resolution.
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ conda remove llvmlite-or1k
Fetching package metadata: ........

Package plan for package removal in environment /home/rabi2/anaconda3:

The following packages will be REMOVED:

    llvmlite-or1k: 0.2.1-py34_0

Proceed ([y]/n)? y

Unlinking packages ...
[      COMPLETE      ]|###################################################| 100%
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ rm -rf $HOME/anaconda3/pkgs/llvmlite-or1k-0.2.1*
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ conda install llvmlite-or1k
Fetching package metadata: ........
Solving package specifications: .
Package plan for installation in environment /home/rabi2/anaconda3:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    llvmlite-or1k-0.2.1        |           py34_1        10.2 MB

The following NEW packages will be INSTALLED:

    llvmlite-or1k: 0.2.1-py34_1

Proceed ([y]/n)? y

Fetching packages ...
llvmlite-or1k- 100% |################################| Time: 0:00:16 664.07 kB/s
Extracting packages ...
[      COMPLETE      ]|###################################################| 100%
Linking packages ...
[      COMPLETE      ]|###################################################| 100%
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ ipython
Python 3.4.3 |Anaconda 2.3.0 (64-bit)| (default, Jun  4 2015, 15:29:08) 
Type "copyright", "credits" or "license" for more information.

IPython 3.2.0 -- An enhanced Interactive Python.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://anaconda.org
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]:  import llvmlite_or1k.binding as llvm

In [2]: exit

```
Does ARTIQ work? no...
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ ls
ddb.pyon  double_ttl_switching.py  foo  pdb.pyon  repository  results  x.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_tests/penning$ artiq_run repository/20150620dds1_v2.py
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 116, in run
    exp_inst = _build_experiment(dmgr, pdb, rdb, args)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 102, in _build_experiment
    return exp(dmgr, pdb, rdb, **arguments)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 45, in __init__
    self.build()
  File "repository/20150620dds1_v2.py", line 6, in build
    self.attr_device("core")
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 80, in attr_device
    setattr(self, key, self.get_device(key))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 77, in get_device
    return self.__dmgr.get(key)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 142, in get
    dev = create_device(desc, self)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 109, in create_device
    module = importlib.import_module(desc["module"])
  File "/home/rabi2/anaconda3/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 16, in <module>
    from artiq.coredevice.runtime import Environment
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/runtime.py", line 3, in <module>
    import llvmlite.ir as ll
ImportError: No module named 'llvmlite'

sbourdeauducq, 2015-07-15:
Update artiq, the latest one definitely imports llvmlite_or1k there and not llvmlite.

closed 2015-07-15 by sbourdeauducq

jboulder, 2015-07-15:
OK. Good to go now. :) 
```bash
$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1242  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
```
Mandelbrot is beautiful.
```bash
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ artiq_run repository/mandelbrot.py 
               ...............................................................
             .....................,,,,,,,,,,,.................................
            .............,,,,,,,,,,,,,,,,,,,,,,,,,,,,,........................
          ..........,,,,,,,,,,,,,,,,,,,,,,,----------,,,,,,...................
         ........,,,,,,,,,,,,,,,,,,,,,------::;h+;::-----,,,,,................
       .......,,,,,,,,,,,,,,,,,,,,,--------:::;+MMhM :------,,,,,.............
      .....,,,,,,,,,,,,,,,,,,,,,---------::::ii+M $hi;::------,,,,,,..........
     ....,,,,,,,,,,,,,,,,,,,,,---------:::;;h      * +;::::----,,,,,,,........
    ....,,,,,,,,,,,,,,,,,,,---------::;;;;ii+$      @hi;;:::::---,,,,,,.......
    ..,,,,,,,,,,,,,,,,,,,-------:::;h+Mh++HHH$      *Hhh+i;;ihi:--,,,,,,,.....
   ..,,,,,,,,,,,,,,,,,------:::::;;i     #               HH M*M+:--,,,,,,,....
  ..,,,,,,,,,,,,,,,,---::::::::;;;i+h#                        h;::-,,,,,,,,...
  .,,,,,,,,,,,,,---:;;::::::;;;;i+                           M+i;:--,,,,,,,,..
 .,,,,,,,,-----::;;M+iiiih+iiii++h                             M :---,,,,,,,,.
 .,,,--------::::;i+$   M# $ HhhH$                             $i;---,,,,,,,,.
 ,,--------:::::;iihM          **                               i:----,,,,,,,,
 ,-------::::;; +hH*                                           h;:----,,,,,,,,
 -:::-:;;;;;ii+h                                              +;::----,,,,,,,,
                                                            Mhi;::----,,,,,,,,
 -:::-:;;;;;ii+h                                              +;::----,,,,,,,,
 ,-------::::;; +hH*                                           h;:----,,,,,,,,
 ,,--------:::::;iihM          **                               i:----,,,,,,,,
 .,,,--------::::;i+$   M# $ HhhH$                             $i;---,,,,,,,,.
 .,,,,,,,,-----::;;M+iiiih+iiii++h                             M :---,,,,,,,,.
  .,,,,,,,,,,,,,---:;;::::::;;;;i+                           M+i;:--,,,,,,,,..
  ..,,,,,,,,,,,,,,,,---::::::::;;;i+h#                        h;::-,,,,,,,,...
   ..,,,,,,,,,,,,,,,,,------:::::;;i     #               HH M*M+:--,,,,,,,....
    ..,,,,,,,,,,,,,,,,,,,-------:::;h+Mh++HHH$      *Hhh+i;;ihi:--,,,,,,,.....
    ....,,,,,,,,,,,,,,,,,,,---------::;;;;ii+$      @hi;;:::::---,,,,,,.......
     ....,,,,,,,,,,,,,,,,,,,,,---------:::;;h      * +;::::----,,,,,,,........
      .....,,,,,,,,,,,,,,,,,,,,,---------::::ii+M $hi;::------,,,,,,..........
       .......,,,,,,,,,,,,,,,,,,,,,--------:::;+MMhM :------,,,,,.............
         ........,,,,,,,,,,,,,,,,,,,,,------::;h+;::-----,,,,,................
          ..........,,,,,,,,,,,,,,,,,,,,,,,----------,,,,,,...................
            .............,,,,,,,,,,,,,,,,,,,,,,,,,,,,,........................
             .....................,,,,,,,,,,,.................................