m-labs/artiq #54: Simultaneous FUD on QC2 DDS hardware

sbourdeauducq, 2015-06-28:
The QC2 (clock lab) DDS hardware has one-hot selection signals, which makes it possible to FUD several channels at the same time.

jboulder, 2015-06-29:
The QC2 DDS hardware is AD9914. Where is this one-hot selection discussed
in the manual? I just checked but didn't see anything.

http://www.analog.com/media/en/technical-documentation/data-sheets/AD9914.pdf

sbourdeauducq, 2015-06-29:
That has to do only with the connections external to the chip.

jboulder, 2015-06-29:
You mean Pin 58 on J0 to J10 of 11-slot motherboard which is routed to
SELECT (pin 19) of SN74CBTLV3245A on the AD9914 daughter cards. The logic
chip is a latch for FUD-IN, RESET-IN, WRB-IN and RDB-IN. OK. I see. That's
convenient for simultaneous update.

sbourdeauducq, 2015-06-29:
It's not exactly what is normally called "latch", but yes.

sbourdeauducq, 2016-10-17:
ARTIQ-Python rewrite of the DDS code is a good occasion for that. Only one-hot selection signals would need to be supported (#576).

sbourdeauducq, 2018-05-13:
There seems to be no interest in this. Closing.

closed 2018-05-13 by sbourdeauducq