m-labs/artiq #57: RTIO loopback test fails on Pipistrello

sbourdeauducq, 2015-07-01:
It worked on the KC705 during the tests we did in the lab 2 weeks ago. It was broken before the ``Quantity`` removal and machine units refactoring.

jordens, 2015-07-01:
http://people.phys.ethz.ch/~robertjo/artiq-ci/artiq-ci-20150628224153-nist_rj1-d2b9400.log
The test output-to-inout (with a wire and two rtios) works but the inout-to-same-inout (same rtio) fails.

sbourdeauducq, 2015-07-05:
Changing unrelated parts of the gateware makes it work. Looks like the typical ISE/Spartan-6 bug.

closed 2015-07-05 by sbourdeauducq

jordens, 2015-07-05:
I think the real issue was the constraints. It did not seem to apply the 8ns constraint in the rtio domain at all. Worked around now in 959ba99f1c9844649fe8dc5144f7ba7a340de300