m-labs/artiq #63: Conda install fails at artiq_flash.sh on clean Xubuntu build

npisenti, 2015-07-03:
I started from a fresh Xubuntu install to get a demo up and running at the JQI, and followed instructions in `doc/manual/installing.rst` (@1d052095ee5143255922319a32110a1dd00f45ec) using `conda`. At the point I go to flash the pipistrello, the bash script errors at line 136. Two comments:

* As of now, the `conda` install does not have a `.../misc/` directory, and so the suggestion to copy in the `udev` rule isn't possible.
* Problem is fixed by manually installing `xc3sprog`, as detailed later in the section to build from source.

Hopefully it's easy to include `xc3sprog` in the conda package to make this workflow easy! Otherwise a hint in the install documentation would be useful... For a while I was convinced it was a problem with the USB device not being seen in my virtualbox image.

fallen, 2015-07-04:
Hi, thanks for reporting these issues :)
I just fixed the first one in https://github.com/m-labs/artiq/commit/813a2a2edf52f182be923704675e13b67758efba
The conda packages for linux-32 and linux-64 have been generated and uploaded to binstar.

About xc3sprog, @sbourdeauducq : do we package it in a separate conda pkg which will be a dependency of the artiq pkg? Or do I just put a link in the manual pointing to where we explain how to build it from source?

sbourdeauducq, 2015-07-04:
Not sure if we should bother too much with xc3sprog - it is shipped with many linux distributions, and replaced later. Making it clear that it should be installed separately is enough.

fallen, 2015-07-04:
Ack

commit 3471ef80 (2015-07-04): manual: closes #63, tell to install xc3sprog before flashing
closed 2015-07-04 by fallen