m-labs/artiq #65: lda: HidError and buffer mixup

jordens, 2015-07-06:
```
======================================================================
ERROR: test_attenuation (artiq.test.lda.TestLda) (i=8.0)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/test/lda.py", line 16, in test_attenuation
    self.assertEqual(i, self.cont.get_attenuation())
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/devices/lda/driver.py", line 205, in get_attenuation
    return (ord(self.get(0x0d, 1))/4.) * dB
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/devices/lda/driver.py", line 186, in get
    self.write(command, length)
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/devices/lda/driver.py", line 155, in write
    len(buf)))
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/devices/lda/driver.py", line 141, in _check_error
    raise HidError("{}: {}".format(ret, err))
artiq.devices.lda.driver.HidError: -1: None

======================================================================
FAIL: test_attenuation (artiq.test.lda.TestLda) (i=8.5)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/test/lda.py", line 16, in test_attenuation
    self.assertEqual(i, self.cont.get_attenuation())
AssertionError: 8.5 != 8.0

======================================================================
FAIL: test_attenuation (artiq.test.lda.TestLda) (i=9.0)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rjordens/src/artiq-testbench/miniconda/envs/artiq/lib/python3.4/site-packages/artiq/test/lda.py", line 16, in test_attenuation
    self.assertEqual(i, self.cont.get_attenuation())
AssertionError: 9.0 != 8.5

```

17:28 < rjo> ysionneau: lda/hidapi looks like it breaks intermittently and then has some queuing
             mixup. any ideas?
http://people.phys.ethz.ch/~robertjo/artiq-ci/artiq-ci_20150705114415-nist_rj1-5db52b1_newassy.log
17:31 < ysionneau> rjo: are those tests running with real hw?
17:35 < ysionneau> ah it seems so
17:35 < ysionneau> I'll have a look tomorrow morning
17:36 < ysionneau> I guess you can open an issue with the first failing test with the Hid exception
17:43 < rjo> yes.
17:44 < rjo> i suspect there are two issues (the error and the queuing mess-up) and both are in hiapi.

jordens, 2015-07-06:
This bug is probably either in the LDA hardware, the USB stack, or hidapi.
@dhslichter You tested and signed of on the LDA driver. Is a workaround (closing the device, reopening, and trying again) good enough?

dhslichter, 2015-07-14:
This workaround seems suitable to me.  The LDA is a "slow" hardware device (so a little extra time to close and reopen is OK from time to time), and has a non-volatile memory so the function of the actual hardware won't be affected by this process.

jordens, 2015-07-14:
Keep in mind that this bug might mean that the device is in a corrupted state and has arbitrary attenuation, at least for a while.

dhslichter, 2015-07-14:
Well, if this is truly due to issues with hidapi we don't have much choice, right?

jordens, 2015-07-14:
Hidapi is open source.

dhslichter, 2015-07-14:
Sure, but it represents some work and then the hidapi maintainers have to want to integrate the suggested changes, because it seems like a bad idea to have our own artiq-friendly version of hidapi that needs to be maintained.  Comments from others?  Obviously it would be nice to have this bug fixed, but I don't know how much work is entailed in tracking it down and writing a fix for hidapi.  Certainly I will not be able to do this.

jordens, 2015-07-14:
Somebody could test whether the bug and/or the workaround have any consequences along the lines I described.

sbourdeauducq, 2019-03-15:
>  it seems like a bad idea to have our own artiq-friendly version of hidapi that needs to be maintained.

Since the upstream hidapi seems to be pretty much abandoned, there isn't much of an alternative if we want this bug fixed.