m-labs/artiq #67: package QC2 FPGA binaries

sbourdeauducq, 2015-07-07:
Ideally we need three sets of FPGA binaries (bitstream/BIOS/runtime): KC705+QC1, KC705+QC2, Pipistrello+QC1. A solution may be to build and package each set separately. If this is excessively complicated, maybe we can stop packaging for Pipistrello for now.

jboulder, 2015-07-14:
We should not abandon Pipistrello builds. Support of this low cost board is a great way of marketing ARTIQ to groups that want to give it a try.

fallen, 2015-08-20:
After some tests, it looks like the entire build can last up to 120 minutes ( https://travis-ci.org/fallen/travis-test/builds/76436045 ) so I think we can add KC705+QC2 without exceeding that time.

fallen, 2015-08-20:
Is there (and will there be) any difference in bios from QC1 builds to QC2 builds?

sbourdeauducq, 2015-08-20:
BIOS no, runtime yes.

fallen, 2015-08-20:
If someone can test with KC705 QC1/QC2 when this build has finished https://travis-ci.org/m-labs/artiq/builds/76490192
To see if artiq_flash.sh -c nist_qc1 and artiq_flash.sh -c nist_qc2 are working correctly or not.

sbourdeauducq, 2015-08-21:
In FMC terminology, "carrier board" is the KC705.

fallen, 2015-08-21:
Ah yes, sorry about that, fixed, thanks!

sbourdeauducq, 2015-08-22:
According to Joe's test, they are not working correctly. Runtime is not flashed:
```bash
rabi2@vboxartiq:~$ artiq_flash.sh -t kc705 -m nist_qc1
Flashing FPGA bitstream...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11412.2 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Created from NCD file: top;UserID=0XFFFFFFFF;Version=2015.1
Target device: 7k325tffg900
Created: 2015/08/21 17:54:38
Bitstream length: 91548896 bits
Erasing sector 175/175. Writing data page  44701/ 44702 at flash page  44701..
Maximum erase time 616.3 ms, Max PP time 61628 us
Verifying page  44702/ 44702 at flash page  44702
Verify: Success!
USB transactions: Write 164591 read 158998 retries 87
Flashing BIOS...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11367.8 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Created from NCD file:
Target device:
Created: 
Bitstream length: 294624 bits
Erasing sector 176/176. Writing data page    143/   144 at flash page  44943..
Maximum erase time 636.7 ms, Max PP time 63673 us
Verifying page    144/   144 at flash page  44944
Verify: Success!
USB transactions: Write 6194 read 601 retries 0
Flashing ARTIQ runtime...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11385.2 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile /home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/binaries/kc705/runtime.fbi: No such file or directory
USB transactions: Write 5602 read 9 retries 0
Done.
rabi2@vboxartiq:~$
```

sbourdeauducq, 2015-08-22:
Remember that it should also be usable as ``artiq_flash.sh -d . -r`` to flash runtime.bin from the current directory (e.g. when doing runtime development)

fallen, 2015-08-26:
It should work now, @jboulder can you test again please?

fallen, 2015-08-28:
I'm closing this, please reopen if there any issue.

closed 2015-08-28 by fallen

jboulder, 2015-08-30:
Raghu has been testing the AD9914 using nist_qc2 configuration. The bit
file programming appears to be working correctly.