m-labs/artiq #68: Support RPCs from core device with constant strings as arguments

sbourdeauducq, 2015-07-10:
Even though the compiler may not support strings, it is desirable to support RPC calls with constant strings in the arguments, so that the upcoming parameter database API can be used directly from the core device (e.g. ``self.set_parameter("parameter_name", value)``).

This can be implemented in the current transforms system by replacing RPC calls containing string arguments with closures that do not take those arguments and inject the strings.

jboulder, 2015-07-17:
Due to #68 the example photon_histogram.py does not work. Please update with example code of how to use the closure trick. This is too cryptic for me to implement: "replacing RPC calls containing string arguments with closures that do not take those arguments and inject the strings."

sbourdeauducq, 2015-07-17:
https://github.com/m-labs/artiq/commit/820ff2da2c69a280f9bb3a031fa10cefc5c28da8 (already referenced above) is an example of how to work around the problem.

sbourdeauducq, 2015-07-17:
And the "closure trick" as you call it is one way to improve the compiler so that such a work around is not necessary.

whitequark, 2015-08-09:
Done in new-py2llvm by passing strings over the wire directly.

jboulder, 2015-08-22:
Here's an example of this work around. 
```python
from artiq import *

def print_underflow():
    print("RTIO underflow occured")

class WhatsWrongHere(EnvExperiment):
    """Should be able to set_result from @kernel decorated method. But this
    is broken with the current compiler. Here's the work around 
    until the new compiler is ready. 
    """

    def set_testval(self, val):
        self.set_result("testval", val, store=False, realtime=True)

    def build(self):
        self.attr_device("core")

    @kernel
    def run(self):
        #self.set_result("test", 0, store=False, realtime=True)
        self.set_testval(0)
```

whitequark, 2015-11-23:
Closing since new compiler was merged.

closed 2015-11-23 by whitequark