m-labs/artiq #70: examples are broken

jboulder, 2015-07-14:
examples/pdb.pyon is missing from github. artiq_run.py errors without it.

And, all the examples are broken. Example below 

```bash
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ artiq_run repository/mandelbrot.py 
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 116, in run
    exp_inst = _build_experiment(dmgr, pdb, rdb, args)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 102, in _build_experiment
    return exp(dmgr, pdb, rdb, **arguments)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 45, in __init__
    self.build()
  File "repository/mandelbrot.py", line 10, in build
    self.attr_device("core")
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 80, in attr_device
    setattr(self, key, self.get_device(key))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 77, in get_device
    return self.__dmgr.get(key)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 142, in get
    dev = create_device(desc, self)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 109, in create_device
    module = importlib.import_module(desc["module"])
  File "/home/rabi2/anaconda3/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 16, in <module>
    from artiq.coredevice.runtime import Environment
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/runtime.py", line 11, in <module>
    llvm.initialize_all_targets()
AttributeError: 'module' object has no attribute 'initialize_all_targets'

sbourdeauducq, 2015-07-14:
It's ``examples/master/ddb.pyon`` and you should run the examples as ``artiq_run repository/xxx.py`` from the master directory. Works fine here.

The other problem is purely LLVM related and should be discussed in issue #50 as I believe this is where the problem is coming from.

closed 2015-07-14 by sbourdeauducq

jboulder, 2015-07-14:
I see #50 is resolved and a new build is finished (build 1228). I've updated but still get the same error.

```bash
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1228  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ artiq_run repository/dds_test.py
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 116, in run
    exp_inst = _build_experiment(dmgr, pdb, rdb, args)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 102, in _build_experiment
    return exp(dmgr, pdb, rdb, **arguments)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 45, in __init__
    self.build()
  File "repository/dds_test.py", line 8, in build
    self.attr_device("core")
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 80, in attr_device
    setattr(self, key, self.get_device(key))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 77, in get_device
    return self.__dmgr.get(key)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 142, in get
    dev = create_device(desc, self)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 109, in create_device
    module = importlib.import_module(desc["module"])
  File "/home/rabi2/anaconda3/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 16, in <module>
    from artiq.coredevice.runtime import Environment
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/runtime.py", line 11, in <module>
    llvm.initialize_all_targets()
AttributeError: 'module' object has no attribute 'initialize_all_targets'

sbourdeauducq, 2015-07-15:
It is not resolved.

sbourdeauducq, 2015-07-15:
And there has been no new build. Packages are still broken in relation to issue #50.

fallen, 2015-07-15:
#50 is reopen, I'll have a look.

fallen, 2015-07-15:
@jboulder could you run the anaconda python interpreter and run : ``import llvmlite_or1k.binding as llvm`` please?