m-labs/artiq #75: artiq_coreconfig error setting idle_kernel

jboulder, 2015-07-16:
Here's the kernel I'd like to run:
```python
from artiq import *

class IdleKernel(EnvExperiment):
    """Idle Kernel"""

    def build(self):
        self.attr_device("core")
        self.attr_device("ttl0")
        self.attr_device("led")
        self.attr_device("sw_doppler")
        self.attr_device("sw_repump")

    @kernel
    def run(self):
        delay(1*ms) 
        while True:
            # cool and state preparation
            self.led.on()
            self.sw_doppler.pulse(5*ms)
            with parallel:
                self.sw_repump.pulse(2*ms)
                self.sw_doppler.pulse(2*ms)
            self.led.off()
            delay(30*ms)

```
Here's the error.
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_coreconfig", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_coreconfig.py", line 48, in main
    comm = create_device(ddb.request("comm"), None)
AttributeError: 'FlatFileDB' object has no attribute 'request'

closed 2015-07-17 by sbourdeauducq

jboulder, 2015-07-17:
Not out of the woods yet on this one. I've also reflashed the FPGA from the latest .bin from binstar.

```bash

rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_gui
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf

KeyboardInterrupt
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ### LED not flashing on Core Device
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/idle.py

KeyboardInterrupt
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ### artiq_run repository/idle.py   <--- This does work
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ 
```
Artiq version
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1252  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$

sbourdeauducq, 2015-07-17:
Can you open a new issue for a different problem?

commit 39c6bc94 (2015-07-17): coreconfig: use new database API (closes #75)