m-labs/artiq #78: PHASE_MODE_ABSOLUTE fails [np19_1251]

jboulder, 2015-07-16:
```python
from artiq import *

class TestDDS(EnvExperiment):
	"""DDS Test PHASE_MODE_ABSOLUTE"""
	def build(self):
		self.attr_device("core")
		self.attr_device("dds_bus")
		self.attr_device("dds1")
		self.attr_device("ttl0")

	@kernel
	def run(self):
		for i in range(200):
			self.ttl0.pulse(2*us)
			self.dds1.set(1*MHz, phase_mode=PHASE_MODE_ABSOLUTE)
			delay(20*ms)

```
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/dds_test_ABSOLUTE.py 
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 243, in code_visit_Name
    obj = self.local_namespace[node.id]
KeyError: 'PHASE_MODE_ABSOLUTE'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 53, in __getitem__
    return self.func_gd[item]
KeyError: 'PHASE_MODE_ABSOLUTE'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 115, in run
    k_function, k_args, k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 101, in compile
    self, k_function, k_args, k_kwargs, with_attr_writeback)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 543, in inline
    kwargs=k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 157, in get_inline
    func_def = func_tr.code_visit(func_def)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 440, in code_visit_FunctionDef
    self.code_generic_visit(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 213, in code_generic_visit
    value = self.code_visit(value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 213, in code_generic_visit
    value = self.code_visit(value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 339, in code_visit_Expr
    self.code_generic_visit(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 223, in code_generic_visit
    new_node = self.code_visit(old_value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 288, in code_visit_Call
    kw.value = self.code_visit(kw.value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 246, in code_visit_Name
    obj = self.global_namespace[node.id]
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 55, in __getitem__
    return getattr(builtins, item)
AttributeError: 'module' object has no attribute 'PHASE_MODE_ABSOLUTE'
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results

sbourdeauducq, 2015-07-17:
Add ``from artiq.coredevice.dds import PHASE_MODE_ABSOLUTE``

closed 2015-07-17 by sbourdeauducq

jboulder, 2015-07-17:
This resolves the original problem:
```
from artiq.coredevice.dds import PHASE_MODE_ABSOLUTE
```
However setting phase doesn't work. I expect the following program to jump the phase of ddsB relative to ddsA. The phase is not touched at all. 
```python
    def test0(self):
        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        self.ddsB.set_phase_mode(PHASE_MODE_ABSOLUTE)
        while True:
            # prepare DDSs
            with self.dds_bus.batch:
                self.ddsA.set(1*MHz, phase=0)
                self.ddsB.set(1*MHz, phase=0)
            delay(5*us)
            # mark start of experiment
            self.ttl0.pulse(2*us)

            self.ttl1.pulse(1*us)
            delay(3*us)
            self.ddsB.set(1*MHz, phase=0.5)
            self.ttl1.pulse(2*us)
            delay(3*us)
            self.ddsB.set(1*MHz, phase=0)
            self.ttl1.pulse(1*us)
            delay(20*ms)

sbourdeauducq, 2015-07-17:
Again, please open a new issue for a different problem.