m-labs/artiq #79: PHASE_MODE_ABSOLUTE fails to set phase

jboulder, 2015-07-17:
Setting phase doesn't work. I expect the following program to jump the phase of ddsB relative to ddsA. The phase is not touched at all. I can change the frequency of ddsB. 

```python
    def test0(self):
        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        self.ddsB.set_phase_mode(PHASE_MODE_ABSOLUTE)
        while True:
            # prepare DDSs
            with self.dds_bus.batch:
                self.ddsA.set(1*MHz, phase=0)
                self.ddsB.set(1*MHz, phase=0)
            delay(5*us)
            # mark start of experiment
            self.ttl0.pulse(2*us)

            self.ttl1.pulse(1*us)
            delay(3*us)
            self.ddsB.set(1*MHz, phase=0.5)
            self.ttl1.pulse(2*us)
            delay(3*us)
            self.ddsB.set(1*MHz, phase=0)
            self.ttl1.pulse(1*us)
            delay(20*ms)

sbourdeauducq, 2015-07-17:
If you have the "prepare DDS" stage only, does it phase-sync them or not?

Please try to have the minimum amount of code necessary to reproduce a problem.

sbourdeauducq, 2015-07-17:
Actually, you should test that with a single DDS and sync a phase jump with a TTL (as done before).

jboulder, 2015-07-28:
Using external trigger... try again. Still not working. See comments at outset of each @kernel.

```python
class TestDDS(EnvExperiment):
    @kernel
    def mdelay(self, d):
        self.ttl1.pulse(d*0.25)
        delay(d*0.75)

    """DDS Test PHASE_MODE_ABSOLUTE"""
    def build(self):
        self.attr_device("core")
        self.attr_device("dds_bus")
        # to switch up which DDSs are used edit ddb.pyon
        self.attr_device("ddsA")
        self.attr_device("ddsB")
        self.attr_device("ddsC")
        self.attr_device("ttl0")
        self.attr_device("ttl1")

    @kernel
    def test5(self):
        """RF phase alignment with logic output"""
        # this works

        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        while True:
            # mark start of experiment
            self.ttl0.pulse(2*us)
            self.ddsA.set(1*MHz, phase=0)
            delay(2.443*ms)  

    @kernel
    def test0(self):
        """absolute phase set at outset; do RF phase increment"""
        # This fails to change phase.

        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        while True:
            # mark start of experiment
            self.ttl0.pulse(2*us)
            self.ddsA.set(2*MHz, phase=0)
            delay(2*us)
            self.ddsA.set(2*MHz, phase=0.5)
            delay(2*us)
            self.ddsA.set(2*MHz, phase=0)
            delay(2*us)
            self.ddsA.set(10*MHz, phase=0)
            delay(1*ms)
    @kernel
    def test4(self):
        """using ddsN.set(freq, phase_mode=PHASE_MODE_ABSOLUTE); do RF phase increment"""
        # This fails to change phase.

        while True:
            # mark start of experiment
            self.ttl0.pulse(2*us)
            self.ddsA.set(2*MHz, phase=0, phase_mode=PHASE_MODE_ABSOLUTE)
            self.mdelay(2*us)
            self.ddsA.set(2*MHz, phase=0.5, phase_mode=PHASE_MODE_ABSOLUTE)
            self.mdelay(2*us)
            self.ddsA.set(2*MHz, phase=0, phase_mode=PHASE_MODE_ABSOLUTE)
            self.mdelay(2*us)
            self.ddsA.set(10*MHz, phase=0, phase_mode=PHASE_MODE_ABSOLUTE)
            delay(1*ms)

    @kernel 
    def test1(self):
        # This fails to change phase.

        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        self.ddsB.set_phase_mode(PHASE_MODE_ABSOLUTE)
        while True:
            self.ttl0.pulse(2*us)
            with self.dds_bus.batch:
                self.ddsA.set(1*MHz, phase=0)
                self.ddsB.set(1*MHz, phase=0)
            self.mdelay(2*us)
            with self.dds_bus.batch:
                self.ddsA.set(1*MHz, phase=0)
                self.ddsB.set(1*MHz, phase=0.5)
            self.mdelay(2*us)
            with self.dds_bus.batch:
                self.ddsA.set(1*MHz, phase=0)
                self.ddsB.set(1*MHz, phase=0)
            delay(1*ms)

sbourdeauducq, 2015-08-27:
Reproduced+fixed on 9914.

jboulder, 2015-08-27:
Still not out of the woods on this one. 

```python
    def test0(self):
        """absolute phase set at outset; do RF phase increment"""
        # This fails to change phase.

        self.ddsA.set_phase_mode(PHASE_MODE_ABSOLUTE)
        #end_delay = 50*us
        end_delay = 53.123*us
        while True:
            # mark start of experiment
            self.ttl0.pulse(2*us)
            self.ddsA.set(2*MHz, phase=0)
            delay(2*us)
            self.ddsA.set(2*MHz, phase=0.5)
            delay(2*us)
            self.ddsA.set(2*MHz, phase=0)
            delay(2*us)
            self.ddsA.set(2*MHz, phase=0)
            delay(end_delay)
```
Using phase locked external references for FPGA and DDSs... 
When end_delay = 50*us the the 0 -> 0.5 and 0.5 -> 0 phase jumps are correct and aligned with ttl0. 
When end_delay = 53.123*us the RF is not phase aligned with ttl0. Rolling.

sbourdeauducq, 2015-08-28:
You need to be very precise and provide more details. Are you 100% sure that:
1. you were seeing the phase jump ("spike" on this oscilloscope trace shortly after the TTL) in the first case
2. adding 3.123us to the end delay completely suppresses the phase jump

jboulder, 2015-08-28:
I can't test now. Next testing opportunity will be Monday.

sbourdeauducq, 2015-09-01:
Testing OK.

commit 0fe0f4d4 (2015-08-27): dds: fix phase computation. Closes #79.
closed 2015-09-01 by sbourdeauducq