m-labs/artiq #80: Specify external RTIO frequency higher than 125MHz for gateware timing analysis

sbourdeauducq, 2015-07-27:
Having ``CLKIN1_PERIOD != CLKIN2_PERIOD`` in the PLL settings messes up the Vivado timing model and router. We are using 8ns for both right now, which limits the external RTIO frequency to 125MHz (SERDES PHY resolution 1ns). Figure out what workarounds to use so that Vivado behaves.

jordens, 2015-07-27:
From playing with the analogous problem on pipistrello, I would suspect that ensuring the less stringent constraint does not take effect (by overwriting or disabling it) might do the trick.

sbourdeauducq, 2017-01-27:
@r-srinivas @dhslichter @dtcallcock Do you plan on using RTIO frequencies higher than 125MHz with the KC705?

klickverbot, 2017-01-27:
@sbourdeauducq: I might be thinking about the wrong clock, but we were thinking about increasing the RTIO frequency to lessen the requirements on the carry-chain length for the in-FPGA TDC. It is not an extremely important consideration, but since propagation on the newer process node is quicker than for your Spartan-6 work, they are quite unwieldy placement-wise.

sbourdeauducq, 2017-01-27:
You could also (and perhaps should) use a local multiplied clock for the TDC.

sbourdeauducq, 2017-01-27:
Yes, and you're not accumulating errors on a long carry chain. But for this sort of high-precision application, note that the Xilinx PLLs aren't particularly good with jitter. You might be better off generating the TDC clock from a low-noise external chip feeding a ``IBUFGDS`` that goes directly to the TDC's sampling FFs.

klickverbot, 2017-01-27:
[This is probably off-topic here anyway, but I believe @cjbe (who actually did the port – I only briefly looked into the options and then used it) also came to that conclusion. In case you ever start integrating a high-res TDC/it gets put on a contract, I would definitely be interested in any progress, though.]

dhslichter, 2017-01-27:
In the NIST KC705 hardware, 125 MHz RTIO clock corresponds to 3 GHz DDS clock.  I don't know of any plans to go to higher frequencies for the DDS clock than this -- the DDS synchronization is not reliable above 3 GHz, and this gives a fine timestamp of 1 ns.  I don't see the need to push to higher RTIO clock frequencies for the time being.

jordens, 2017-01-27:
For SAWG every bit of faster RTIO frequency helps in terms of lower resource usage, wider bandwidth, and better signal quality.

sbourdeauducq, 2017-01-27:
The problem in the issue at hand arises from clock input switching (internal and external clocks with a constraint already existing on the internal and a tighter one desired on the external). The SAWG design is not affected.

closed 2017-01-27 by sbourdeauducq