m-labs/artiq #81: idle_kernel does not automatically run

jboulder, 2015-07-27:
This is related to #75. 

```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat repository/idle.py
from artiq import *

class IdleKernel(EnvExperiment):
    """Idle Kernel"""

    def build(self):
        self.attr_device("core")
        self.attr_device("ttl0")
        self.attr_device("led")
        self.attr_device("sw_doppler")
        self.attr_device("sw_repump")

    @kernel
    def run(self):
        delay(1*ms) 
        while True:
            # cool and state preparation
            self.led.on()
            self.sw_doppler.pulse(5*ms)
            with parallel:
                self.sw_repump.pulse(2*ms)
                self.sw_doppler.pulse(2*ms)
            self.led.off()
            delay(100*ms)
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ## No LED blinking... The idle experiment ought to be running automatically at this point.
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/idle.py 
^CTraceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.serve(rpc_map, exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 226, in serve
    _, ty = self._read_header()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
    (c, ) = struct.unpack("B", self.read(1))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_tcp.py", line 32, in read
    rn = self.socket.recv(min(8192, length - len(r)))
KeyboardInterrupt
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ## LED blinking. Press CTRL-C to break. LED continues to blink for 2 seconds. Then LED stops blinking.
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/ttl_test.py 
^CTraceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.serve(rpc_map, exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 226, in serve
    _, ty = self._read_header()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
    (c, ) = struct.unpack("B", self.read(1))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_tcp.py", line 32, in read
    rn = self.socket.recv(min(8192, length - len(r)))
KeyboardInterrupt
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ## NO LED blinking at any point during or after this run.
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1264  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0

sbourdeauducq, 2015-07-27:
Did you upgrade and try again? I fixed that problem in aba2d3f112ffbd0bfea1657ed535a95c05fc57a7 and tested it OK on my KC705 board.

jboulder, 2015-07-27:
The connection between github commit ID and Binstar build number is opaque
to me. It's not at either of these URLs. Should I be looking elsewhere?

https://binstar.org/fallen/artiq/files
https://anaconda.org/fallen/artiq/files

jboulder, 2015-07-27:
I updated conda. Reflashed FPGA. Problem persists. 
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1265  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results

rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat repository/idle.py
```
```python
from artiq import *

class IdleKernel(EnvExperiment):
    """Idle Kernel"""

    def build(self):
        self.attr_device("core")
        self.attr_device("ttl0")
        self.attr_device("led")
        self.attr_device("sw_doppler")
        self.attr_device("sw_repump")

    @kernel
    def run(self):
        delay(1*ms) 
        while True:
            # cool and state preparation
            self.led.on()
            self.sw_doppler.pulse(5*ms)
            with parallel:
                self.sw_repump.pulse(2*ms)
                self.sw_doppler.pulse(2*ms)
            self.led.off()
            delay(100*ms)
```
```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle.py
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/idle.py 
^CTraceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.serve(rpc_map, exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 226, in serve
    _, ty = self._read_header()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
    (c, ) = struct.unpack("B", self.read(1))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_tcp.py", line 32, in read
    rn = self.socket.recv(min(8192, length - len(r)))
KeyboardInterrupt

```
LED blinking. Press CTRL-C to break. LED continues to blink for 2 seconds. Then LED stops blinking. Stays solid illuminated. 

Repeated this sequence twice to same effect.

sbourdeauducq, 2015-07-28:
Cannot reproduce on my board. The LED never stops blinking here, which indicates the experiment is correctly run alternatively from flash and from the PC control.

What are your DDB entries for the devices? Also, can you try with a simpler test case that just blinks a LED and does not touch other devices?

jboulder, 2015-07-30:
Problem persists with a simpler idle program. 

```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ 
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ pwd
/media/sf_virtualBoxSharedJake/britton/artiq_penning
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ conda update artiq
Fetching package metadata: ........
# All requested packages already installed.
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                   np19_1265  
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ ls
ddb.pyon  foo  pdb.pyon  repository  results
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat repository/idle_simple.py
from artiq import *
from artiq import *

class IdleKernel(EnvExperiment):
    """Idle Kernel"""

    def build(self):
        self.attr_device("core")
        self.attr_device("led")

    @kernel
    def run(self):
        delay(1*ms) 
        while True:
            self.led.on()
            delay(100*ms)
            self.led.off()
            delay(100*ms)
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_compile repository/idle_simple.py 
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig write -f idle_kernel repository/idle_simple.elf
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ # no blinking -- core did not automatically start... this is a bug
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_run repository/idle_simple.py 
^CTraceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.serve(rpc_map, exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 226, in serve
    _, ty = self._read_header()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
    (c, ) = struct.unpack("B", self.read(1))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_tcp.py", line 32, in read
    rn = self.socket.recv(min(8192, length - len(r)))
KeyboardInterrupt
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ # after CTRL-C led continued to blink for several seconds then halted at bright
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ 
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ 
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat pyon.ddb
cat: pyon.ddb: No such file or directory
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat 
ddb.pyon    foo         pdb.pyon    repository/ results/    
```

```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ cat ddb.pyon
# This is configured for the Penning lab's KC705
{
    "comm": {
        "type": "local",
[... 186 lines cut ...]
        "arguments": {"sysclk": 1e9,
                      "channel": 7}
    },

    "pmt": "pmt0",
    "dds_nsf": "dds0", # nuclear spin flip
    "dds_odf_lower": "dds4",
    "dds_odf_upper": "dds5",
    "exp_start": "ttl0",
    "sw_doppler": "ttl1",
    "sw_repump": "ttl2",
    "sw_uw": "ttl5",
    "sw_odf_lower": "ttl6",
    "sw_odf_upper": "ttl9",
    "sw_pmt1_andor0": "ttl12",
    #names for testing of DDSs using dds_tests.py
    "ddsA": "dds1",
    "ddsB": "dds7",
    "ddsC": "dds3"
}
```

```bash
rabi2@vboxartiq:/media/sf_virtualBoxSharedJake/britton/artiq_penning$ artiq_coreconfig read idle_kernel
b'\x7fELF\x01\x02\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\\\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\xc8\x00\x00\x00\x00\x004\x00\x00\x00\x00\x00(\x00\n\x00\x07\xd7\xe1O\xfc\x9c!\xff\xe4\xd4\x01\xd0\x14\xd4\x01\xc0\x10\xd4\x01\xb0\x0c\xd4\x01\xa0\x08\xd4\x01\x90\x04\x04\x00\x00\x00\xd4\x01p\x00\xaaK\x00\x00\x04\x00\x00\x00\xaa\x8c\x00\x00\x04\x00\x00\x00\xa8k\x00\x00\xbc+\x00\x00\x10\x00\x004\x15\x00\x00\x00\x18`\x00\x01\xa8\x83\xe8H\xe1\xd4 \x00\x9c`\x00\x01\xe4\x8e\xa0\x00\x10\x00\x00\x03\xa8\xa3\x00\x00\x18\xa0\x00\x00\xe4\x8e \x00\x10\x00\x00\x03\x15\x00\x00\x00\xa8e\x00\x00\xe2R\x18\x00\x18`\x00\xbe\xab\x03\xbc \x18`\x01}\xabCx@\x00\x00\x00\x04\x9e\xc0\x00\x11\xe2R\xa0\x00\xa9\xc3\x00\x00\x9e\x80\x00\x01\xa8r\x00\x00\xa8\x8e\x00\x00\xa8\xb6\x00\x00\x04\x00\x00\x00\xa8\xd4\x00\x00\xe0\x8e\xc0\x00\xe4\x84p\x00\x10\x00\x00\x03\xa8t\x00\x00\x18`\x00\x00\xe4\x84\xc0\x00\x10\x00\x00\x03\xa8\xb4\x00\x00\xa8\xa3\x00\x00\xe0r(\x00\xa8\xb6\x00\x00\x04\x00\x00\x00\x18\xc0\x00\x00\xe0n\xd0\x00\xe4\x83p\x00\x10\x00\x00\x03\xa8\x94\x00\x00\x18\x80\x00\x00\xe4\x83\xd0\x00\x13\xff\xff\xe5\x15\x00\x00\x00\x03\xff\xff\xe3\xaa\x84\x00\x00\x04\x00\x00\x00\x15\x00\x00\x00\xa9\xcb\x00\x00\xa8r\x00\x00\x04\x00\x00\x00\xa8\x94\x00\x00\x04\x00\x00\x00\xa8n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01zR\x00 [... cut ...] _setjmp\x00__syscall_ttl_set_o\x00__eh_getid\x00__syscall_now_save\x00__eh_raise\x00\x00\x00\x00\x1c\x00\x00\x0b\x06\x00\x00\x00\x00\x00\x00\x00(\x00\x00\x08\x06\x00\x00\x00\x00\x00\x00\x000\x00\x00\n\x06\x00\x00\x00\x00\x00\x00\x00\xa8\x00\x00\r\x06\x00\x00\x00\x00\x00\x00\x00\xdc\x00\x00\r\x06\x00\x00\x00\x00\x00\x00\x01\x0c\x00\x00\x07\x06\x00\x00\x00\x00\x00\x00\x01\x1c\x00\x00\x0c\x06\x00\x00\x00\x00\x00\x00\x01$\x00\x00\t\x06\x00\x00\x00\x00\x00\x00\x00\x1c\x00\x00\x02\x01\x00\x00\x00\x00'

sbourdeauducq, 2015-07-31:
Cannot reproduce, the ELF file recovered from your artiq_coreconfig read works correctly as idle kernel on my board.

sbourdeauducq, 2015-07-31:
Bah. There is no travis build integrating my fix. I thought you had checked that...

sbourdeauducq, 2015-07-31:
There is a package with the fix now, you may test and reopen if the problem still exists.

closed 2015-07-31 by sbourdeauducq

jboulder, 2015-07-31:
Confirmed that this works on NIST Penning lab's KC705. Required reflash of
FGPA. Using conda build artiq 0.0 np_1357.