m-labs/artiq #86: New ARTIQ Python compiler checklist

whitequark, 2015-08-09:
Currently the new compiler misses many features at edges, often subtle. This issue is intended to track them to ensure none are forgotten.

  - [ ] `Fraction`
  - [ ] RPC return type annotations
  - [ ] `syscall`
  - [ ] classes and methods
  - [ ] embedding of foreign objects
  - [ ] emission of debug info via LLVM
  - [ ] symbolization of core device backtraces
  - [ ] interleaving

sbourdeauducq, 2015-08-10:
Interleaving, classes and syscalls are by far the highest priority as users cannot do anything with the new compiler until this is done.

We don't need to keep compatibility with ``artiq.language.core.int64``. Removing it is good.

whitequark, 2015-11-23:
Merged.

closed 2015-11-23 by whitequark