m-labs/artiq #88: Ubuntu 14.04 dependencies missing for git support

jboulder, 2015-08-12:
sudo apt-get install libhttp-parser2.1 libssh2-1

sbourdeauducq, 2015-08-13:
In general, libraries are installed automatically as dependencies of other packages. What package requires those, exactly?

fallen, 2015-08-13:
Those are dependencies of libgit2 , I will have a look at if I can either package those in conda or somehow link libgit2 statically with them at build time

fallen, 2015-08-13:
seems that it's very likely you already have libssh2-1. Since git, gitk, cmake, gnupg, libreoffice, vlc for instance require it

fallen, 2015-08-13:
I've just re-uploaded the libgit2 package, removing the dependency on libhttp-parser.
It still depends on libssh2 though, but this should be installed on most distros, is it really not installed on Ubuntu 14.04?

jboulder, 2015-08-13:
My experience yesterday on Xubuntu 14.04 was that both these libraries
needed to be installed via apt-get.

fallen, 2015-08-14:
So : I removed the dependency on libhttp-parser and I packaged libssh2.
It should be fixed now.

closed 2015-08-14 by fallen