m-labs/artiq #90: git repository and  ddb.pyon

jboulder, 2015-08-12:
artiq_master needs to be launched in a folder containing ddb.pyon and pdb.pyon. Following is the typical directory structure for ARTIQ. 
```bash
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ ls
ddb.pyon  pdb.pyon  results  testrepo
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ ls -la testrepo/
total 48
drwxr-xr-x 3 rabi2 rabi2 4096 Aug 12 15:59 .
drwxr-xr-x 4 rabi2 rabi2 4096 Aug 12 16:08 ..
-rw-r--r-- 1 rabi2 rabi2  721 Aug 12 15:59 arguments_demo.py

rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ artiq_master -g -r testrepo
```
Here, testrepo is cloned from a local repository I setup on my machine. I can launch artiq_master and see the experiments in artiq_gui (labeled by git commit ID). 

Unfortunately, ddb.pyon and pdb.pyon are not included in the repository. They need to be since they're a crucial part of reproducing the experiment state if a user wants to roll-back. I'll try to add them to the git-controlled repository and restart artiq_master in the new location. .
```bash
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ ls
ddb.pyon  pdb.pyon  results  testrepo
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ cp *.pyon testrepo
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master$ cd testrepo
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master/testrepo$ ls
arguments_demo.py  flopping_f_simulation.py  pdb.pyon             tdr.py
ddb.pyon           handover.py               photon_histogram.py  transport.py
dds_test.py        mandelbrot.py             speed_benchmark.py
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master/testrepo$ git add *.pyon
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master/testrepo$ git commit
[master 0f99eef] add .pyon
 2 files changed, 146 insertions(+)
 create mode 100644 ddb.pyon
 create mode 100644 pdb.pyon
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master/testrepo$ git push
No refs in common and none specified; doing nothing.
Perhaps you should specify a branch such as 'master'.
fatal: The remote end hung up unexpectedly
error: failed to push some refs to '/home/rabi2/testrepo'
rabi2@vboxartiq:~/artiq-dev/artiq/examples/master/testrepo$ artiq_master -g -r .
```
Not sure why I get this git error but I guess pushing isn't required. Anyway, artiq_master can be successfully started. Note this begs a different default ARTIQ experiment folder hierarchy. Now the results directory is intermixed with the program files. Do we want this?

sbourdeauducq, 2015-08-13:
Git integration of PDB and DDB is not supported and requires significant work.

I'm not sure about the PDB, but I would argue that this is not a feature we want for the DDB. The DDB does not contain information that changes the behavior of an experiment, but just what device is connected where in the system. If using an older revision of the experiment also reverts to an older version of the DDB, this would require the user to shuffle devices around (e.g. swap DDS channels) whereas keeping the current DDB configuration would typically run the older experiment immediately and correctly.

(Parameter history will be kept in InfluxDB, not Git)

jboulder, 2015-08-14:
For your visit to Boulder I'll add a TODO for discussing if we want to make PDB and DDB part of the git maintained source.

jboulder, 2015-08-19:
> If using an older revision of the experiment also reverts to an older version of the DDB, this would
> require the user to examine the DDB changes and shuffle devices around (e.g. swap DDS channels) 
> whereas keeping the current DDB configuration would typically run the older experiment immediately 
> and correctly.

The PDB may be needed for experiment replay. A record of changes to DDB are helpful as it helps user ascertain if old experiment can be rerun verbatim or if hardware/DDB changes are needed. Seems like this could be as easy as including *.pyon in git repository.

sbourdeauducq, 2015-08-19:
No, it's **not** as easy as including the pyon files in git. git is not magical and won't make the master read the new pyon files, handle multiple versions of them when experiments are run in parallel with different versions of the PDB, do the right thing when an experiment modifies a parameter, etc.

jboulder, 2015-08-19:
Agreed that having multiple simultaneous versions of PDB is confusing. But
master needs to be restarted anyway upon PDB edit.... We can discuss when
you guys arrive.

sbourdeauducq, 2015-08-19:
It's not confusing, it's just complex and needs a proper solution, not only "add to Git and not worry about it".
Parameters can be changed without master restart from the console in the GUI or from the command line client.

sbourdeauducq, 2015-10-04:
As discussed during the last visit, a good solution for DDB versioning is to put a copy of ddb.pyon into the repository without the master doing anything special with it. The user should manually diff/update/overwrite the ddb.pyon used by the master as needed, and then have the master re-scan it.

ddb.pyon can now be rescanned without a master restart by using ``artiq_client scan-ddb``.

closed 2015-10-04 by sbourdeauducq