m-labs/artiq #92: binstar 64-bit Windows build missing

jboulder, 2015-08-13:
The m-labs website [advertises](http://m-labs.hk/artiq/) support for Microsoft Windows. However, only 32-bit binaries are being built by [binstar](https://binstar.org/fallen/artiq). 

* This is not made clear on the ARTIQ home page or the installation instructions.
http://m-labs.hk/artiq/manual/installing.html
http://m-labs.hk/artiq/manual/installing.html

* Most modern Windows installations are 64-bit. I advocate that ARTIQ should include a 64-bit build for Window 7.

jboulder, 2015-08-13:
Note this is a codification of one of the topics discussed in #64.

jordens, 2015-08-15:
You have a skewed notion of "support". Packages are not required to support something. They are a mere convenience. Put (64-bit or not) windows at the bottom of the list unless somebody stops piling up unstructured wishlist items and steps up to volunteer for sustained testing.

Justin318, 2015-08-21:
If you guys don't have so much troubles compiling a version for the 64-bit windows machine, it would be convenient to have one.  Most of the people in the lab will still try to run artiq using a 64-bit windows and will appreciate the ease of use.

dleibrandt, 2015-08-21:
I agree.  I would use it on 64 bit windows if it were available.

tingrei86, 2015-08-22:
I agree. With artiq compatible with windows x64, we could possibly attract other groups' interest in using artiq.

jordens, 2015-08-22:
Good. Would you four volunteer to do testing then? The testing and the signoff is happening a bit unorganized now. We are not keeping track of who tests what. Anybody willing to organize that?

And since the windows builds are manual and time consuming, we should focus on WoW64 and dump 32.

Again, there is nothing that I can see that would prevent using the 32 bit packages on 64 bit windows. They are definitely compatible with 64 bit windows and should run fine on a 64 bit machine. So most of the reasons given above should be void.

jboulder, 2015-08-22:
See #91 for a discussion of what testing and sign-off might entail.

jordens, 2015-08-24:
The only two reasons that I can think of that would need 64 bit packages (most of artiq should be noarch anyway so this is a non-issue) are more memory (big number crunching) and drivers where we use ctypes that do not have a 32 bit userspace. pydaqmx may be of the later case and even while itself can be noarch it would pull in a 64 bit runtime.

dleibrandt, 2015-08-24:
To modify my previous comment: I don’t necessarily think that we need to have 64 bit artiq binaries (I don’t need big number crunching), but it would be nice if the supplied 32 bit artiq binaries could run on 64 bit windows.  If that is already the case, than please ignore my comment.

fallen, 2015-08-26:
You should use win-32 packages on 64-bit Windows (you must therefore uninstall 64-bit Anaconda/miniconda and install the 32-bit version) : https://github.com/m-labs/artiq/commit/7b0b1d686013691fb5a492b3691715c6611cf2ab

closed 2015-08-26 by fallen