m-labs/artiq #100: pause() broken

jboulder, 2015-08-14:
I expect adding [pause()](http://m-labs.hk/artiq/manual/management_system.html#experiment-scheduling) permit yieldings. It looks like this is broken. Or am I doing something wrong? 
```python
from artiq import *

class LED(EnvExperiment):
    def build(self):
        self.attr_device("core")
        self.attr_device("led")
        self.attr_device("scheduler")

    @kernel
    def run(self):
        dt = 500*ms
        while True: 
            self.led.on()
            delay(dt)
            self.led.off()
            delay(dt)
            self.scheduler.pause()
```
```bash
rabi2@vboxartiq:~/myartiq$ artiq_run myartiqrepo/led_blink.py 
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 115, in run
    k_function, k_args, k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 101, in compile
    self, k_function, k_args, k_kwargs, with_attr_writeback)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 543, in inline
    kwargs=k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 157, in get_inline
    func_def = func_tr.code_visit(func_def)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 440, in code_visit_FunctionDef
    self.code_generic_visit(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 213, in code_generic_visit
    value = self.code_visit(value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 213, in code_generic_visit
    value = self.code_visit(value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 339, in code_visit_Expr
    self.code_generic_visit(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 223, in code_generic_visit
    new_node = self.code_visit(old_value)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 197, in code_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 285, in code_visit_Call
    func = self.static_visit(node.func)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 446, in static_visit
    return visitor(node)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/transforms/inline.py", line 463, in static_visit_Attribute
    return getattr(value, node.attr)
AttributeError: 'DummyScheduler' object has no attribute 'pause'
```
Problem appears to be in worker_impl.py
```python
class Scheduler:
    pause = staticmethod(make_parent_action("pause", ""))
```
Also, documentation is not explicit about how to call pause.

sbourdeauducq, 2015-08-15:
Obviously you need a scheduler (and a master) to use ``pause``. ``artiq_run`` doesn't have a scheduler, it's just a stand-alone experiment running tool. ``worker_impl`` is not used by ``artiq_run``.

sbourdeauducq, 2015-08-15:
As for the doc: http://m-labs.hk/artiq/manual/management_system.html#pauses

closed 2015-08-15 by sbourdeauducq

jboulder, 2015-08-15:
> Obviously you need a scheduler (and a master) to use pause. 
> artiq_run doesn't have a scheduler, it's just a stand-alone 
> single-shot experiment running tool. worker_impl is not used 
> by artiq_run.
Indeed. My bad. But we found a tiny bug. :) 

Just confirmed that when using artiq_gui to schedule experiments that things work fine. This is after I add self.scheduler.pause() to the above program.

commit 34a9c8c1 (2015-08-15): artiq_run: add dummy pause method (closes #100)