m-labs/artiq #93: choose whne to regenerate GUI

jboulder, 2015-08-14:
This Issue was first discussed in #89.

It's rather helpful to be able to force reloading (or not) of default values for an experiment. For example, at present Scannable supports min and max but not default_min and default_max. Syntactically we call this default lower bound, dlb in the current gui. So the scan options are: dlb, dub, lb and ub. 

Here's an example of how we use this. When conducting an experiment to measure the center of a resonance the sweep range depends on the pi-time t_pi and last known f0, these are Ion Parameters. The default value for the sweep range is a function of Ion Property t_pi and last known Ion Property f0. The sweep range might be manually overridden via GUI  to something other than default as a diagnostic. The sweep range in the GUI may also reflect outdated values for f0 and t_pi. 

In our present HFgui a user can choose whether or not to reload (and recompute) the default sweep range. It may be that in ARTIQ a checkbox would work.

```python 
self.attr_argument("reload_default_attr", BooleanValue(True), default=False) 
```  

But there needs to be a way of telling the GUI about this choice.

sbourdeauducq, 2015-08-15:
How does the GUI tell "the sweep range" from the other arguments for which you want to keep the values the user entered?

jboulder, 2015-08-15:
I believe the following would reproduce the functionality of our existing HFgui. 

Each attr_ has an default parameter which can be an anonymous function. 
Each attr_ has an bool override parameter. The state of the override is indicated in the GUI by a checkbox on the far right side. Implementation follows.

When program enters prepare() phase, for each attr_
if override=True:
    make so that run() uses attr_ cached by GUI
if override = False: 
    attr_ default is evaluated, GUI is updated and, run() uses new configuration 

When the GUI override state transitions from True to False:
     attr_ default is evaluated, GUI is updated

sbourdeauducq, 2015-12-11:
I have implemented something equivalent.

closed 2015-12-11 by sbourdeauducq