m-labs/artiq #94: needed: error message if core device is missing

jboulder, 2015-08-14:
Report communication failure with core device at two locations.

* on command line for artiq_main even if no verbose flags are set
* in artiq_gui log

sbourdeauducq, 2015-08-15:
It is already reported in artiq_gui log. What is artiq_main?

jboulder, 2015-08-15:
Sorry, I meant artiq_master not artiq_main.

Regardless, I can unplug or power off the Core Device and there are no messages that appear in artiq_gui's log. 

I observe the scheduler can be put into an inconsistent state.
* with the Core Device attached start a program
* disconnect Core Device
* in artiq_gui tell scheduler to delete program
* it keeps running on the Core Device
* reconnect Core Device
* deleted program continues to run even though the scheduler things it has been stopped

sbourdeauducq, 2015-08-15:
Upgrade.

sbourdeauducq, 2015-08-15:
Also, don't expect any error message to appear about the core device being disconnected when nothing is using it. This is not an error.

jboulder, 2015-08-15:
>Also, don't expect any error message to appear about the core device being disconnected when >nothing is using it. This is not an error.

But I expect that the following is undesirable behavior. Right? 

>    with the Core Device attached start a program
>    disconnect Core Device
>    in artiq_gui tell scheduler to delete program
>    it keeps running on the Core Device
>    reconnect Core Device
>    deleted program continues to run even though the scheduler things it has been stopped

Looks like binstar may be stuck again... Last build was 2 days ago. 
OK. Waiting for next binstar build... It looks like its stuck again...

sbourdeauducq, 2015-08-15:
https://binstar.org/m-labs

closed 2015-08-15 by sbourdeauducq

fallen, 2015-08-15:
Just FYI, after 31st of August binstar.org will be deprecated, everybody should use anaconda.org -> https://anaconda.org/m-labs/.
Maybe we should say officially that my own anaconda repository is not to be used any longer?
@jboulder I uploaded all the useful old packages from fallen to m-labs account FYI.