m-labs/artiq #95: ddb.pyon documentation

jboulder, 2015-08-14:
Document arguments for different modules.
```python
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {"ref_period": 1e-9}
    },
    "pmt0": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLInOut",
        "arguments": {"channel": 0}
    },
    "ttl0": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 2}
    },
    "dds0": {        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9858",
        "arguments": {"sysclk": 1e9, "channel": 0}
    },
```
General ARTIQ Devices 
```python
    "lda": {
        "type": "controller",
        "best_effort": true,
        "host": "::1",
        "port": 3253,
        "target_name": "lda",
        "command": "lda_controller -p {port} --bind {bind} --simulation"
    },
```
Composite devices need special discussion.
```python
    "qc_q1_0": {
        "type": "controller",
        "best_effort": false,
        "host": "::1",
        "port": 4000,
        "target_name": "pdq2",
        "command": "pdq2_controller -p {port} --bind {bind} --simulation --dump qc_q1_0.bin"
    },
    "electrodes": {
        "type": "local",
        "module": "artiq.devices.pdq2",
        "class": "CompoundPDQ2",
        "arguments": {
            "pdq2_devices": ["qc_q1_0", "qc_q1_1", "qc_q1_2", "qc_q1_3"],
            "trigger_device": "ttl3",
            "frame_devices": ["ttl0", "ttl1", "ttl2"]
        }
    },
```
Document what presently appears at the end of a file.
```python
    "ttl_inout": "pmt0",
    "ttl_out": "ttl0",
    "ttl_out_serdes": "ttl0",

    "pmt": "pmt0",
    "bd_dds": "dds0",
    "bd_sw": "ttl0",
    "bdd_dds": "dds1",
    "bdd_sw": "ttl1"
```

jordens, 2015-08-15:
Pasting together random stuff that others wrote does not constitute an issue and is not helpful. Why don't you RTFM about the NDSPs and then come back with a refined item list?

sbourdeauducq, 2015-08-18:
http://m-labs.hk/artiq/manual/environment.html

closed 2015-08-18 by sbourdeauducq

jboulder, 2015-08-18:
The fact that I didn't supply a patch constituting a solution to this issue
is off topic. Please reopen this issue.

The ARTIQ documentation needs to provide users with enough information that
they can figure out how to configure their ddb.pyon. What's [here](
http://m-labs.hk/artiq/manual/environment.html#ddb) isn't enough.

For example, optional key value pairs for the Core device include
ref_period and external_clock. This is an opportunity to document that the
default is ref_period=8*ns and that the RTIO resolution is ref_period/8.
What's the minimum ref_period for KC705? How does user specify internal vs
external clock?

DDS sysclk must be 8*ref_period for AD9858. Core Device reference clock
must be phase locked to sysclk. channel values depend on system
configuration.

TTL... tell users that the channel value to pin mapping are on the
following page
http://m-labs.hk/artiq/manual/core_device.html?highlight=kc705

Tell users that other devices are [NDSP] (
http://m-labs.hk/artiq/manual/developing_a_ndsp.html?highlight=ndsp). That
NDSP command line options are discussed [here] (
http://m-labs.hk/artiq/manual/ndsp_reference.html). And that there's a
relationship between command line options and ddb.pyon syntax.

sbourdeauducq, 2015-08-19:
Most of those points are about the core device, not DDB.

About the last point, the doc already says "Controller managers run the command field in a shell to launch the controller".

jboulder, 2015-08-19:
Some core device features are configured in ddb.pyon. Presently, the
documentation of what constitutes ddb.pyon is insufficient. How, for
example would I know this is the location within ARTIQ the core device is
configured to use IP vs USB?

sbourdeauducq, 2015-08-19:
Serial-over-USB has limitations that makes it somewhat unsuitable for end users.