m-labs/artiq #96: artiq_run broken; `GLIBCXX_3.4.20' not found

jboulder, 2015-08-14:
```bash
rabi2@vboxartiq:~/myartiq$ cat myartiqrepo/led_blink.py 
```
```python
from artiq import *

class LED(EnvExperiment):
    def build(self):
        self.attr_device("core")
        self.attr_device("led")

    @kernel
    def run(self):
        dt = '100*ms'
        while True:
            self.core.break_realtime()
            self.led.on()
            delay(dt)
            self.led.off()
            delay(dt)
```
```bash
rabi2@vboxartiq:~/myartiq$ 
rabi2@vboxartiq:~/myartiq$ 
rabi2@vboxartiq:~/myartiq$ artiq_run myartiqrepo/led_blink.py
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 116, in run
    exp_inst = _build_experiment(dmgr, pdb, rdb, args)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 102, in _build_experiment
    return exp(dmgr, pdb, rdb, **arguments)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 118, in __init__
    self.build()
  File "myartiqrepo/led_blink.py", line 6, in build
    self.attr_device("core")
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 177, in attr_device
    setattr(self, key, self.get_device(key))
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/environment.py", line 172, in get_device
    return self.__dmgr.get(key)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 151, in get
    dev = _create_device(desc, self)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/master/worker_db.py", line 118, in _create_device
    module = importlib.import_module(desc["module"])
  File "/home/rabi2/anaconda3/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 16, in <module>
    from artiq.coredevice.runtime import Runtime
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/runtime.py", line 4, in <module>
    import llvmlite_artiq.binding as llvm
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/llvmlite_artiq/binding/__init__.py", line 6, in <module>
    from .dylib import *
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/llvmlite_artiq/binding/dylib.py", line 4, in <module>
    from . import ffi
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/llvmlite_artiq/binding/ffi.py", line 36, in <module>
    lib = ctypes.CDLL(os.path.join(_lib_dir, get_library_name()))
  File "/home/rabi2/anaconda3/lib/python3.4/ctypes/__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /home/rabi2/anaconda3/lib/python3.4/site-packages/llvmlite_artiq/binding/libllvmlite.so)
```

fallen, 2015-08-14:
You can fix this by doing:
`sudo add-apt-repository ppa:ubuntu-toolchain-r/test`
`sudo apt-get update`
`sudo apt-get install libstdc++6`

I will add this to the manual.

closed 2015-08-14 by fallen

fallen, 2015-08-14:
Done there https://github.com/m-labs/artiq/commit/8669f1d61db7cacc676db464ffbcd54b76e3e8e0

jboulder, 2015-08-14:
> Done there 8669f1d

Confirmed that this resolves the bug .

fallen, 2015-08-17:
I rebuilt the llvmlite-artiq package for linux-64 on an Ubuntu 14.04 VM, so the workaround is no longer necessary, I am deleting it from the manual.

jboulder, 2015-08-17:
Awesome! Thanks.