m-labs/artiq #97: check that FPGA has latest firmware

jboulder, 2015-08-14:
On occasions modifications to ARTIQ require compilation of a new bit file. Upon startup artiq_master should check to see if the Core Device is up to date. If not prompt user to run artiq_flash.sh.

sbourdeauducq, 2015-08-15:
This is complicated (due to the problem of tracking which bitfiles changes are incompatible and which are not). How important is that?

jordens, 2015-08-15:
I don't think we have to do any meaningful tracking and versioning of the gateware interface/runtime API yet. As outlined on the mailing list, having the host-side code query the embedded gateware git hash, the bios git hash and the runtime git hash and compare against its own would go a long way. Then reduce the mismatch error from something fatal to a warning if the `pdb` entry for the core device has `ignore_mismatch` which you could use during development.
I suspect `versioneer` might come handy here.
Low priority IMHO. Feel free to label as such.

sbourdeauducq, 2015-11-24:
Let me take care of this one, will be an occasion to touch the new comms code.

commit eeb53c5a (2015-12-11): coredevice/comm: compare software and gateware versions. Closes #97
closed 2015-12-11 by sbourdeauducq