m-labs/artiq #99: RTIOSequenceError(RuntimeException) due to self.core.break_realtime()

jboulder, 2015-08-14:
The following runs fine.
```python
class LED(EnvExperiment):
    def build(self):
        self.attr_device("core")
        self.attr_device("led")

    @kernel
    def run(self):
        dt = 500*ms
        while True: 
            # self.core.break_realtime()
            self.led.on()
            delay(dt)
            self.led.off()
            delay(dt)
```
Uncommenting self.core.break_realtime() results in the following error. 
```bash
rabi2@vboxartiq:~/myartiq$ artiq_run myartiqrepo/led_blink.py 
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/bin/artiq_run", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 133, in main
    return run(with_file=True)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/frontend/artiq_run.py", line 118, in run
    exp_inst.run()
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/language/core.py", line 115, in run_on_core
    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.serve(rpc_map, exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 230, in serve
    self._serve_exception(rpc_wrapper, user_exception_map)
  File "/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq/coredevice/comm_generic.py", line 218, in _serve_exception
    raise exception(self.core, p0, p1, p2)
artiq.coredevice.runtime_exceptions.RTIOSequenceError: at 0.0 on channel 0
```
This is also different behavior than expected from RTIOSequenceError(RuntimeException)
```python
class RTIOSequenceError(RuntimeException):
    """Raised when an event is submitted on a given channel with a timestamp
    not larger than the previous one.
    The offending event is discarded and the RTIO core keeps operating.
    """
```

sbourdeauducq, 2015-08-15:
What happens here is you are programming a LED off event far in the future, then set ``now`` to a value close to the RTIO counter, then program a LED on event before the LED off you had already programmed. ``led.sync()`` at the end of the loop would fix this.

closed 2015-08-15 by sbourdeauducq

jboulder, 2015-08-19:
Shouldn't the RTIO core keep operating for RTIOSequenceError? To what does "channel 0" refer in the exception? If led, that's [channel 18](http://m-labs.hk/artiq/manual/core_device.html)...

sbourdeauducq, 2015-08-19:
It keeps operating in the sense that other events are not affected (or are they?). Exception parameters are broken due to issue #30.