m-labs/artiq #104: Dock Area crash upon artiq_gui restart

tingrei86, 2015-08-21:
I am encountering problems of GUI docking location does not restored to the last state upon restarting. I noticed this problem occurs after I undock certain widgets from the main dock before restarting.

The error message I received was: artiq_gui.state: Failed to restore state for object 'DockArea'

Upon multiple restart, sometimes certain widgets are missing, this issues is not repeatable upon multiple restart.

sbourdeauducq, 2015-08-22:
Can you send a artiq_gui.pyon that exhibits the problem?

sbourdeauducq, 2015-08-22:
See also: pyqtgraph/pyqtgraph#204 - but ARTIQ should not run into this bug.

jboulder, 2015-08-31:
```
{
    "ResultsDock": {"yy": {
        "ty": "XY",
        "state": {
            "gridAlphaSlider": 128,
            "pointsGroup": true,
            "averageGroup": false,
            "xGridCheck": false,
            "paramList": {},
            "forgetTracesCheck": false,
            "fftCheck": false,
            "downsampleCheck": false,
            "autoAlphaCheck": false,
            "yGridCheck": false,
            "subsampleRadio": false,
            "clipToViewCheck": false,
            "logXCheck": false,
            "logYCheck": false,
            "view": {
                "xInverted": false,
                "targetRange": [[npscalar("float64", "AAAAAAAA4L8="), npscalar("float64", "AAAAAAAA4D8=")], [npscalar("float64", "AAAAAAAA4L8="), npscalar("float64", "AAAAAAAA4D8=")]],
                "background": null,
                "linkedViews": [null, null],
                "wheelScaleFactor": -0.125,
                "mouseMode": 3,
                "enableMenu": true,
                "mouseEnabled": [true, true],
                "yInverted": false,
                "viewRange": [[npscalar("float64", "AAAAAAAA4L8="), npscalar("float64", "AAAAAAAA4D8=")], [npscalar("float64", "AAAAAAAA4L8="), npscalar("float64", "AAAAAAAA4D8=")]],
                "autoPan": [false, false],
                "autoRange": [true, true],
                "autoVisibleOnly": [false, false],
                "aspectLocked": false,
                "limits": {
                    "yRange": [null, null],
                    "xRange": [null, null],
                    "xLimits": [null, null],
                    "yLimits": [null, null]
                }
            },
            "alphaSlider": 1000,
            "peakRadio": true,
            "meanRadio": false,
            "maxTracesSpin": 0,
            "autoDownsampleCheck": true,
            "maxTracesCheck": false,
            "downsampleSpin": 1,
            "autoPointsCheck": true,
            "alphaGroup": true
        },
        "settings": {
            "x": "<None>",
            "error": "<None>",
            "y": "common_plot_x",
            "fit": "<None>"
        }
    }},
    "MainWindow": b'\x01\xd9\xd0\xcb\x00\x01\x00\x00\x00\x00\x00f\x00\x00\x00\xc7\x00\x00\x03\xe9\x00\x00\x04\xd3\x00\x00\x00g\x00\x00\x00\xdf\x00\x00\x03\xe8\x00\x00\x04\xd2\x00\x00\x00\x00\x00\x00',
    "ExplorerDock": {},
    "DockArea": {
        "main": ("tab", [], {"index": -1}),
        "float": [({
            "main": ("vertical", [], {"sizes": []}),
            "float": []
        }, (463, 214, 640, 176))]
    }
}

```

jboulder, 2015-10-22:
I encounter the same problem frequently. Here's an example.
```
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ ./start.bash 
discarding /home/jwbritto/anaconda3/envs/py35/bin from PATH
prepending /home/jwbritto/anaconda3/envs/py35/bin to PATH
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ WARNING:ctlmgr(68810artiq2):artiq.frontend.artiq_ctlmgr:Connection to master failed (ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 3250))
WARNING:ctlmgr(68810artiq2):artiq.frontend.artiq_ctlmgr:Retrying in 5.0 seconds
WARNING:artiq.frontend.artiq_influxdb:Connection to master failed (ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 3250))
WARNING:artiq.frontend.artiq_influxdb:Retrying in 5.0 seconds
WARNING:artiq.gui.datasets:Failed to create display 'default_bright_mean_strip'
Traceback (most recent call last):
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/datasets.py", line 159, in restore_state
    desc["settings"])
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/datasets.py", line 136, in create_display
    self.update_display_data(dsp)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/datasets.py", line 104, in update_display_data
    for k in dsp.data_sources()})
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/datasets.py", line 104, in <dictcomp>
    for k in dsp.data_sources()})
KeyError: 'default_bright_mean_strip_x'
WARNING:artiq.gui.datasets:Failed to restore display state of 'default_bright_mean_strip'
Traceback (most recent call last):
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/datasets.py", line 164, in restore_state
    dsp.restore_state(desc["state"])
UnboundLocalError: local variable 'dsp' referenced before assignment
WARNING:artiq.gui.state:Failed to restore state for object 'DockArea'
Traceback (most recent call last):
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/state.py", line 57, in load
    _restore_state(obj, state)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/state.py", line 24, in _restore_state
    method(state)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/pyqtgraph/dockarea/DockArea.py", line 242, in restoreState
    self.moveDock(d, 'below', None)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/pyqtgraph/dockarea/DockArea.py", line 115, in moveDock
    self.addDock(dock, position, neighbor)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/pyqtgraph/dockarea/DockArea.py", line 104, in addDock
    old.apoptose()
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/pyqtgraph/dockarea/Container.py", line 61, in apoptose
    self.container().insert(self.widget(0), 'before', self)
AttributeError: 'NoneType' object has no attribute 'insert'
```
```
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ cat artiq_gui.pyon 
{
    "LogDock": {"min_level_idx": 0},
    "MainWindow": b'\x01\xd9\xd0\xcb\x00\x01\x00\x00\x00\x00\x00\xaf\x00\x00\x01\xf9\x00\x00\x06\xa2\x00\x00\x04\x8e\x00\x00\x00\xaf\x00\x00\x01\xf9\x00\x00\x06\xa2\x00\x00\x04\x8e\x00\x00\x00\x00\x00\x00',
    "ExplorerDock": {
        "default": {
            "expanded": [],
            "argument_values": {
                "cool_t": 0.003,
                "hist_nmax": 200,
                "nreps": 100,
                "detect_t": 0.001,
                "no_cool_t": 0.03,
                "simulation_mode": false,
                "hist_nbins": 60
            }
        },
        "Novatech Wall": {
            "expanded": [],
            "argument_values": {
                "dfdt_max": 1000.0,
                "reset_novatech": false,
                "reset_dds_phases": false,
                "wall_order_m": "2",
                "gain": 0.0,
                "scan": {
                    "npoints": 10,
                    "ty": "LinearScan",
[... 94 lines cut ...]
                "enableMenu": true,
                "background": null,
                "viewRange": [[0, 1], [0, 1]],
                "yInverted": false,
                "linkedViews": [null, null],
                "mouseMode": 3,
                "autoPan": [false, false],
                "xInverted": false,
                "wheelScaleFactor": -0.125
            },
            "pointsGroup": true,
            "gridAlphaSlider": 128,
            "autoAlphaCheck": false,
            "peakRadio": true,
            "autoDownsampleCheck": true,
            "logYCheck": false,
            "clipToViewCheck": false,
            "averageGroup": false,
            "autoPointsCheck": true,
            "yGridCheck": false
        }
    }}
}

```

sbourdeauducq, 2016-02-14:
Should not be a problem anymore with QDockWidget.

closed 2016-02-14 by sbourdeauducq