m-labs/artiq #106: Scannable

Justin318, 2015-08-22:
I was testing the GUI using the template "flopping_f_simulation.py" in the examples folder and realized there maybe some problems with the function Scannable. In the following, the scan method "Linear" has been chosen in the GUI unless specifically mentioned. 

1. A "Scannable" in the GUI has the attributes "min" and "max", although the value assigned to min can be larger than the value of max. In this case, the can starts from the larger value to a smaller value. This is not really a problem and even a feature for some cases, since one may want to scan in the reversed order sometimes. But the attribute here means rather a start and a stop instead of min and max. 

2. A "Scannable" has the attribute global_min and global_max, which are from my understanding limiting  the minimum and maximum of the scan varriable. With the global_min = 1000, global_max=4000 set in the script "flopping_f_simulation", the attribute min in the GUI in deed limited between 1000 and 4000, but the attribute max can change between 1000 and 4043.00. This seems like a indexing issue in the code. 

3. If one does not set the attribute global_min and global_max for a Scannable in the script, but only the default value 
self.attr_argument("frequency_scan", Scannable(defualt=LinearScan(1000, 4000, 100)))
The attribute min for the Scannable is limited in the GUI to 1030 and the attribute max is limited up to 4043, which I don't think is a wished behavior. The lower bound for both min and max turned out to be 0.0. But in principle if not global bounds are set, the variable should not be limited at all. 
 
4. Precision. Is there a way to choose the precision of digits for the parameters "min" and "max". It seems at the moment, they are limited to two digits.

sbourdeauducq, 2015-08-22:
I cannot reproduce item 2. Can you post the full source?

sbourdeauducq, 2015-08-22:
Regarding item 3: the standard Qt widgets do not allow spinboxes without min/max. There is a solution in Pyqtgraph, but as often: pyqtgraph/pyqtgraph#210

sbourdeauducq, 2015-08-22:
Since we may want a fancier scan widget later, supporting that min > max corner case is problematic. 284e3ddb2b5e19eef2cffb0b671aae0b7bb2582f now enforces min <= max.

sbourdeauducq, 2015-08-24:
Seems Qt groks ``QDoubleSpinbox.setMaximum(float("inf"))`` and same with min/-inf. It is a rare event that one of those IEEE754 special cases produces something else than bugs.

d5f2f5c062e77a2f09bb6ab44b2cd4dc9607ba3c and e04317912068797364b9bacf3ab75f94706f99aa should fix the rest of those issues.

closed 2015-08-24 by sbourdeauducq

tingrei86, 2015-08-26:
So, how do we increase the precision to more than 2 digits after decimal point?

sbourdeauducq, 2015-08-26:
Have you looked at the commits I referenced above?