m-labs/artiq #107: Artiq_run command hangs when FPGA is not turned on

amhankin, 2015-08-25:
When entering the command, "artiq_run led.py" using the example program "led.py" hangs when the FPGA is turned off. The expected behavior is for the program to timeout and throw an error message after failed communication for a set amount of time.

jboulder, 2015-08-25:
https://github.com/m-labs/artiq/issues/94

I've seen similar behavior.

jordens, 2015-08-25:
That is correct behavior. It aligns with how CLI tools operate in general. Use `timeout 10s artiq_run led.py`. But detecting and handling core device hang/disconnect/off in the master is something we need to define.

sbourdeauducq, 2015-08-26:
Is that problem on windows or linux?

jordens, 2015-08-26:
Does tcp keepalive do anything if the connection is not (yet) alive?

sbourdeauducq, 2015-08-26:
No, but I was assuming the core device was turned off *after* the connection, which is something TCP without keepalive has a problem dealing with. Attempting to connect to a host that is down results pretty quickly in a "No route to host" error because ARP times out, or a similar error, without any particular TCP option.

@amhankin So what do you mean by "hang"?

amhankin, 2015-08-26:
@sbourdeauducq 
I was using a linux installation. By "hang", I mean the program runs without ever returning to a standard mode that allows for user input. 

A simple "Ctrl-C" will interrupt the program and allow you to continue trying other things, but the lack of feedback from the program can be confusing when you are attempting to communicate with an FPGA for the first time (not a new user but I have not used artiq for over four months).

sbourdeauducq, 2015-08-26:
How long are you waiting for before assuming it hangs?

amhankin, 2015-08-26:
I would say somewhere around 10 to 20 seconds.

sbourdeauducq, 2015-08-26:
And what happens after?

sbourdeauducq, 2015-08-28:
I have added another 5-second connect timeout in addition to the regular TCP error handling mechanisms. It is reasonable to assume that if your network cannot establish a TCP connection in 5 seconds, you would not be able to do anything useful later on with the core device either.

closed 2015-08-28 by sbourdeauducq