m-labs/artiq #108: AD9914 amplitude control

raghavendrasrinivas, 2015-08-25:
Is there any way to control the amplitude from the AD9914? The manual describes phase and frequency control but not amplitude.

sbourdeauducq, 2015-08-27:
What should the amplitude unit be?

raghavendrasrinivas, 2015-08-27:
Could we just leave it as a fraction?

sbourdeauducq, 2015-08-27:
You mean a number between 0 (min) and 1 (max)?

raghavendrasrinivas, 2015-08-27:
Yes. I think that's probably better than any absolute power or voltage value as most of the time the output is amplified anyway.

whitequark, 2015-08-28:
@sbourdeauducq What do you want me to do?

dhslichter, 2015-08-28:
Needs to be a new method which takes an amplitude between 0 and 1, converts to a 12-bit value, and writes to ASF register.

raghavendrasrinivas, 2015-09-04:
diff --git a/artiq/coredevice/dds.py b/artiq/coredevice/dds.py
index e9e8303..9759ec2 100644
--- a/artiq/coredevice/dds.py
+++ b/artiq/coredevice/dds.py
@@ -89,6 +89,11 @@ class _DDSGeneric:
         word."""
         return pow/2**self.pow_width
 
+    @portable
+    def amplitude_to_asf(self, amplitude):
+        """Converts amplitude from scale of 0 to 1 to 0 to 2^12."""
+        return round(amplitude*0x0fff)
+
     @kernel
     def init(self):
         """Resets and initializes the DDS channel.
@@ -117,12 +122,13 @@ class _DDSGeneric:
         self.phase_mode = phase_mode
 
     @kernel
-    def set_mu(self, frequency, phase=0, phase_mode=_PHASE_MODE_DEFAULT):
+    def set_mu(self, frequency, phase=0, phase_mode=_PHASE_MODE_DEFAULT, amplitude=0x0fff):
         """Sets the DDS channel to the specified frequency and phase.
 
         This uses machine units (FTW and POW). The frequency tuning word width
         is 32, whereas the phase offset word width depends on the type of DDS
-        chip and can be retrieved via the ``pow_width`` attribute.
+        chip and can be retrieved via the ``pow_width`` attribute. The amplitude
+        width is 12.
 
         :param frequency: frequency to generate.
         :param phase: adds an offset, in turns, to the phase.
@@ -132,13 +138,14 @@ class _DDSGeneric:
         if phase_mode == _PHASE_MODE_DEFAULT:
             phase_mode = self.phase_mode
         syscall("dds_set", now_mu(), self.channel, frequency,
-                phase, phase_mode)
+                phase, phase_mode, amplitude)
 
     @kernel
-    def set(self, frequency, phase=0.0, phase_mode=_PHASE_MODE_DEFAULT):
+    def set(self, frequency, phase=0.0, phase_mode=_PHASE_MODE_DEFAULT,
+            amplitude=1.0):
         """Like ``set_mu``, but uses Hz and turns."""
         self.set_mu(self.frequency_to_ftw(frequency),
-                    self.turns_to_pow(phase), phase_mode)
+                    self.turns_to_pow(phase), phase_mode, self.amplitude_to_asf(amplitude))
 
 
 class AD9858(_DDSGeneric):
diff --git a/artiq/coredevice/runtime.py b/artiq/coredevice/runtime.py
index ae818b1..5fcaf56 100644
--- a/artiq/coredevice/runtime.py
+++ b/artiq/coredevice/runtime.py
@@ -25,7 +25,7 @@ _syscalls = {
     "dds_init": "Ii:n",
     "dds_batch_enter": "I:n",
     "dds_batch_exit": "n:n",
-    "dds_set": "Iiiii:n",
+    "dds_set": "Iiiiii:n",
 }
 
 
diff --git a/examples/master/ddb.pyon b/examples/master/ddb.pyon
index cafe5e5..443bba6 100644
--- a/examples/master/ddb.pyon
+++ b/examples/master/ddb.pyon
@@ -9,7 +9,7 @@
         "type": "local",
         "module": "artiq.coredevice.core",
[... 109 lines cut ...]
         if(batch_count >= DDS_MAX_BATCH)
@@ -210,9 +218,10 @@ void dds_set(long long int timestamp, int channel,
         batch[batch_count].ftw = ftw;
         batch[batch_count].pow = pow;
         batch[batch_count].phase_mode = phase_mode;
+        batch[batch_count].amplitude = amplitude;
         batch_count++;
     } else {
         rtio_chan_sel_write(RTIO_DDS_CHANNEL);
-        dds_set_one(timestamp - DURATION_PROGRAM, timestamp, channel, ftw, pow, phase_mode);
+        dds_set_one(timestamp - DURATION_PROGRAM, timestamp, channel, ftw, pow, phase_mode, amplitude)
     }
 }
diff --git a/soc/runtime/dds.h b/soc/runtime/dds.h
index 03c2bf0..2e3a4d0 100644
--- a/soc/runtime/dds.h
+++ b/soc/runtime/dds.h
@@ -59,6 +59,6 @@ void dds_init(long long int timestamp, int channel);
 void dds_batch_enter(long long int timestamp);
 void dds_batch_exit(void);
 void dds_set(long long int timestamp, int channel,
-    unsigned int ftw, unsigned int pow, int phase_mode);
+    unsigned int ftw, unsigned int pow, int phase_mode, unsigned int amplitude);
 
 #endif /* __DDS_H */

fallen, 2015-09-04:
You can post this as a gist https://gist.github.com/ or on http://pastebin.com/

sbourdeauducq, 2015-09-04:
Actually, better send those patches to the ML or by email to me.
There are still some issues to fix, e.g. this disables reset for the 9858 as well, adds the unneeded reset pulse delays, etc. Please review carefully each line of this patch.

raghavendrasrinivas, 2015-09-04:
Added asf_to_amplitude function
http://pastebin.com/RVLji9xQ

sbourdeauducq, 2015-09-04:
Almost there, there is still the reset test laying around in this patch, and the ddb should not be modified.

raghavendrasrinivas, 2015-09-05:
I think I got rid of all the residual reset test stuff.

http://pastebin.com/1vpURX2F

sbourdeauducq, 2015-09-05:
The "turns on OSK" comment is misplaced. DURATION_DAC_CAL should not be modified.

raghavendrasrinivas, 2015-09-05:
We're not changing the DAC cal pulse time to match the datasheet? We don't want it at 1 ms?

sbourdeauducq, 2015-09-05:
147000 is already the correct value for 1+ms, I already committed the fix. In all cases, this should be in a separate patch, since it has nothing to do with amplitude control.

raghavendrasrinivas, 2015-09-05:
Okay. Where do you want the turn on OSK comment? That seems to be where we modified it.

sbourdeauducq, 2015-09-05:
The comment and the source code currently sound as if bit 0x2000 were OSK disable.

raghavendrasrinivas, 2015-09-05:
Is this clearer?
http://pastebin.com/2nax8WE0

sbourdeauducq, 2015-09-05:
Yes, thanks

commit b3831d33 (2015-09-05): dds: support amplitude tuning on AD9914

Closes #108
closed 2015-09-05 by sbourdeauducq