m-labs/artiq #111: LED doesn't work using nist_qc2

raghavendrasrinivas, 2015-08-25:
I tried flashing the KC705 with nist_qc2 but the led.py example doesn't seem to work. I reflashed it to the default and the LED works again. Is this an issue with the TTL channel?

jboulder, 2015-08-26:
Related, how does soc/runtime choose AD9914 or AD9858? Can see #ifdef DDS_AD9914 but nowhere #define DDS_AD9914.

sbourdeauducq, 2015-08-26:
The RTIO channel numbers are different on QC1 and QC2. Did you update your DDB?

jboulder, 2015-08-26:
They're missing in the docs.

http://www.m-labs.hk/artiq/manual/core_device.html#kc705

It's unclear to me which pins correspond to user_led.

artiq
<https://github.com/m-labs/artiq/tree/2c15bd3e44086de30e5a5f5fe1db9780fc1c12c2>
/soc
<https://github.com/m-labs/artiq/tree/2c15bd3e44086de30e5a5f5fe1db9780fc1c12c2/soc>
/targets
<https://github.com/m-labs/artiq/tree/2c15bd3e44086de30e5a5f5fe1db9780fc1c12c2/soc/targets>
/*artiq_kc705.py*

sbourdeauducq, 2015-08-26:
The problem is it's not clear yet what the QC2 hardware will end up being and I do not want to spend my time refactoring docs. So RTFS for now.

sbourdeauducq, 2015-08-26:
LED works fine here. The correct channel number is 16.

closed 2015-08-26 by sbourdeauducq

jboulder, 2015-08-26:
Which file and line number should I be looking at?

sbourdeauducq, 2015-08-26:
artiq_kc705.py, the channel numbers are the indices in the rtio_channels list.