m-labs/artiq #112: artiq_flash.sh, runtime.fbi error

jboulder, 2015-08-26:
Looks like runtime.fbi isn't included in the latest conda 1493. Conversation moved to github for tracking. 

>It is in fact included, but now instead of being in binaries/kc705
>it is in binaries/kc705/nist_qc1 (or /nist_qc2).
>From now on we build both nist_qc1 and nist_qc2 FPGA bitstream (and runtime).
>You can chose which mezzanine board you want to use when using artiq_flash.sh by 
>providing the -m parameter (see the help)

At present you **must** choose which mezzanine board you want.
However, the script --help says that the default is nist_qc1. Please
make the script execution reflect this default.

That problem aside it still doesn't work right. As of conda 1502. 

```bash
(root)rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq$
artiq_flash.sh -t kc705 -m nist_qc1
Flashing FPGA bitstream...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent
Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11206.4 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/media/sf_git/m-labs/artiq/artiq/binaries/kc705/artiq_kc705-nist_qc1-kc705.bit:
No such file or directory
USB transactions: Write 5602 read 9 retries 0
Flashing BIOS...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent
Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11213.5 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/media/sf_git/m-labs/artiq/artiq/binaries/kc705/bios.bin: No such file
or directory
USB transactions: Write 5602 read 9 retries 0
Flashing ARTIQ runtime...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent
Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11191.3 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/media/sf_git/m-labs/artiq/artiq/binaries/kc705/nist_qc1/runtime.fbi:
No such file or directory
USB transactions: Write 5602 read 9 retries 0
Done.

(root)rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq$ conda list
artiq# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                     np_1502
cairoplot3-artiq          3.1.2                    py34_0
gbulb-artiq               0.1                      py34_0
llvmlite-artiq            0.5.1                    py34_2
```

fallen, 2015-08-26:
About the default not working for -m it is very weird because it should already work: https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_flash.sh#L13

About the flashing issue, you have conflicts between your locally installed artiq module (installed by manually typing python3 setup install or python3 setup develop) and your artiq conda package.
Make sure your environment is correctly using all the conda installation:
`$ which python` should output the path to anaconda's python binary.
`$ python -c "import artiq; print(artiq.__path__[0])"` should print the path to artiq module **within** anaconda environment

jboulder, 2015-08-26:
```bash
$ which python/home/rabi2/anaconda3/bin/python
/home/rabi2/anaconda3/bin/python

$ python -c "import artiq; print(artiq.__path__[0])"
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq
```

fallen, 2015-08-26:
Ah, the `artiq_flash.sh` script uses "python3", can you retype those 2 commands with `python3` instead of `python`?

jboulder, 2015-08-26:
```
$ which python3
/home/rabi2/anaconda3/bin/python3
$ python3 -c "import artiq; print(artiq.__path__[0])"
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq
```

fallen, 2015-08-26:
Hmmm this does not make any sense, can you open the file at `which artiq_flash.sh` and uncomment line 6 which should be "set -x" https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_flash.sh#L6 and run it again please?

jboulder, 2015-08-26:
Edited. Rerun.

```bash
rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq/exps/diagnostic$
artiq_flash.sh -t kc705 -m nist_qc1
++ python3 -c 'import artiq; print(artiq.__path__[0])'
+
ARTIQ_PREFIX=/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq
+ BOARD=kc705
+ MEZZANINE_BOARD=nist_qc1
+ getopts bBrht:d:f:m: opt
+ case $opt in
+ '[' kc705 == kc705 ']'
+ BOARD=kc705
+ getopts bBrht:d:f:m: opt
+ case $opt in
+ '[' nist_qc1 == nist_qc1 ']'
+ MEZZANINE_BOARD=nist_qc1
+ getopts bBrht:d:f:m: opt
+ '[' -z '' ']'
+ '[' kc705 == kc705 ']'
+ UDEV_RULES=99-kc705.rules
+ BITSTREAM=artiq_kc705-nist_qc1-kc705.bit
+ CABLE=jtaghs1_fast
+ PROXY=bscan_spi_kc705.bit
+ BIOS_ADDR=0xaf0000
+ RUNTIME_ADDR=0xb00000
+ RUNTIME_FILE=nist_qc1/runtime.fbi
+ FS_ADDR=0xb40000
+ '[' -z '' ']'
+
BIN_PREFIX=/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705
+ search_for_proxy bscan_spi_kc705.bit
+ PROXY=bscan_spi_kc705.bit
+ '[' -f /home/rabi2/.migen/bscan_spi_kc705.bit ']'
+ PROXY_PATH=/home/rabi2/.migen/
+ '[' -z '' -a -z '' -a -z '' -a -z '' ']'
+ FLASH_RUNTIME=1
+ FLASH_BIOS=1
+ FLASH_BITSTREAM=1
+ set +e
+ xc3sprog -c jtaghs1_fast -R
+ '[' 0 '!=' 0 ']'
+ set -e
+ '[' '!' -z '' ']'
+ '[' 1 == 1 ']'
+ echo 'Flashing FPGA bitstream...'
Flashing FPGA bitstream...
+ xc3sprog -v -c jtaghs1_fast -I/home/rabi2/.migen//bscan_spi_kc705.bit
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/artiq_kc705-nist_qc1-kc705.bit:w:0x0:BIT
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB
Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11331.7 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/artiq_kc705-nist_qc1-kc705.bit:
No such file or directory
USB transactions: Write 5602 read 9 retries 0
+ '[' 1 == 1 ']'
+ echo 'Flashing BIOS...'
Flashing BIOS...
+ xc3sprog -v -c jtaghs1_fast -I/home/rabi2/.migen//bscan_spi_kc705.bit
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/bios.bin:w:0xaf0000:BIN
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB
Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11304.5 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/bios.bin:
No such file or directory
USB transactions: Write 5602 read 9 retries 0
+ '[' 1 == 1 ']'
+ echo 'Flashing ARTIQ runtime...'
Flashing ARTIQ runtime...
+ xc3sprog -v -c jtaghs1_fast -I/home/rabi2/.migen//bscan_spi_kc705.bit
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/nist_qc1/runtime.fbi:w:0xb00000:BIN
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
    http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
    http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB
Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093    Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/02/26 13:03:52
Bitstream length: 91548896 bits
done. Programming time 11303.3 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000
256 bytes/page, 65536 pages = 16777216 bytes total
Can't open datafile
/home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq/binaries/kc705/nist_qc1/runtime.fbi:
No such file or directory
USB transactions: Write 5602 read 9 retries 0
+ echo Done.
Done.
+ xc3sprog -v -c jtaghs1_fast -R
```

jboulder, 2015-08-26:
Part of the problem is that the binaries directory doesn't exist.
```bash
$ ls /home/rabi2/anaconda3/lib/python3.4/site-packages/artiq-0.0+dev-py3.4.egg/artiq
coredevice  gateware     language   py2llvm      test        wavesynth
devices     gui          master     __pycache__  tools.py
frontend    __init__.py  protocols  sim          transforms
rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq/exps/diagnostic$ conda list anaconda
# packages in environment at /home/rabi2/anaconda3:
#
anaconda                  2.3.0                np19py34_0  
rabi2@vboxartiq:/media/sf_git/nistpenning/calc/artiq/exps/diagnostic$ conda list artiq
# packages in environment at /home/rabi2/anaconda3:
#
artiq                     0.0                     np_1502  
cairoplot3-artiq          3.1.2                    py34_0  
gbulb-artiq               0.1                      py34_0  
llvmlite-artiq            0.5.1                    py34_2  
```

fallen, 2015-08-26:
This path is not the installed conda ARTIQ package.
The conda ARTIQ package installs in there: `lib/python3.4/site-packages/artiq/`
I think you must have done some `python setup.py install` while in the anaconda environment.
Please remove this .egg file

jboulder, 2015-08-26:
OK. Looks like this fixed the problem!

So, going forward is there a way to make sure that artiq_flash.sh is looking in the right place for its dependencies?

closed 2015-08-26 by sbourdeauducq