m-labs/artiq #115: dds write function

raghavendrasrinivas, 2015-08-27:
Would it be difficult to implement a function that lets you write specific values to the DDS registers? This is sometimes useful for troubleshooting the DDS' or trying out new features on it. I'm thinking of sometimes along the lines of dds.write(address, values).

jboulder, 2015-08-27:
I propose m-labs give a brief tutorial at NIST next week on how to add this
feature to ARTIQ. This would touch the ARTIQ codebase at a number of levels
and would be instrumental in teaching us how to implement features like
this ourselves. I'm sure Robert already knows how to do this so the
audience is more others in Ion Storage.

I guess it would entail editing the runtime and ARTIQ-runtime bindings.

https://github.com/m-labs/artiq/tree/master/soc/runtime

-Joe

sbourdeauducq, 2015-08-28:
Have you tried the runtime test mode (press "t" during boot, while connected to the serial console)?

raghavendrasrinivas, 2015-08-28:
Yes. But sometimes it's useful if we can toggle the sync_clk on and off within the experiment. For example to test how long the DDS's can stay synchronised, or to see how the internal clock is affected by certain events.

dhslichter, 2015-08-28:
I can definitely envision use cases where we would want to re-run the DAC cal, for example, or redo the syncing to ensure it hasn't slipped, immediately before running an experiment.  I would suggest refactoring to create simple register read and write methods, which are then called by the higher level functions (set freq/amplitude/etc).  The preferred way to do things would be to use the high-level functions, but users who need the functionality could use the low-level to have arbitrary register control as needed.  This would also mean putting all the different register addresses into dds.h

In addition, once ARTIQ supports SPI we plan to be programming registers of other AD9914 chips over SPI, including multiple profiles and/or digital ramp modes. Such things would be "hand-coded", in other words the ARTIQ compiler would not be expected to keep track of DDS state (it would be up to the person writing the experiment), but an arbitrary register read/write function would be the tool to use there.

sbourdeauducq, 2015-08-29:
1. As far as I know, the syncing will be done continuously by the hardware planned at the moment so the DDS should not lose sync.
2. For fiddling with the DDS, you can reboot into test mode (this doesn't reset the DDS) and from there do things like disabling sync_clk and reading/writing registers. Is the DDS so unreliable that you need to meddle with it amid experiments? I can imagine this is the case with the SCSI adapters and no syncing, but hopefully the new hardware will improve that significantly.
3. You can re-run the DAC calibration already by using ``dds.init``. Anyway, does ADI recommend periodic DAC re-calibration? The datasheet only mentions it must be done once after power-up, leaving power-down modes, and changing the clock. It also mentions it is about meeting internal digital timing, not about analog circuitry.
4. How do profiles and digital ramp modes depend on SPI?
5. ``dds.h``, like all other runtime includes, is not accessible to experiments.
6. My understanding is one of the points of ARTIQ is to raise the level of abstraction and improve portability. Writing directly to the DDS registers from the experiment goes against that goal.

dhslichter, 2015-08-31:
1. The DDS syncing hardware, when run continuously, should not lose sync, but it would be nice to be able to do this in case we run into issues when pushing the clock frequency higher.  We have seen the syncing circuit sometimes lock to one of two different phases, and having the ability to just restart the syncing process might enable us to fix that.
2/3. The DDS is not totally unreliable, but for example one might have duty cycle issues where a DAC cal performed upon startup (with a cold DDS) should be redone later on once the DDS chips have come to temperature, or if duty cycles (and thus DDS dissipation) are changing then the temp and thus calibration might shift.  These are real considerations for many of our experiments.  If dds.init() performs this task, then it should be OK.
4. They don't depend on SPI.  However, once SPI is running in artiq, we will be programming a different set of DDS chips over SPI from the core device, and these chips we will want to access all the registers because they will use profile and digital ramp modes.  The profiles and ramps will be hand-coded for the time being, since implementing them in the compiler raises complexity.   
6. Yes, we want to raise the level of abstraction and thus portability, but this should be balanced against the ability to have fine-grained control if need be.  I think that having methods for dds.set_register() or dds.get_register() would both allow the existing code to be nicely refactored and would give the opportunity for fine-grained control with a low-level command when needed.  One would obviously encourage the use of the high-level dds methods unless the low-level ones are necessary.

jordens, 2016-07-20:
@r-srinivas @dhslichter  Is this still desired or resolved if #364 is merged?

sbourdeauducq, 2016-08-15:
I guess not; reopen if needed.

closed 2016-08-15 by sbourdeauducq