m-labs/artiq #116: delay between dds.init()

raghavendrasrinivas, 2015-08-27:
It seems like one needs a delay of 500 us between dds.init commands for different dds'. If not, there is an RTIO sequence error. I tried sending init commands through the dds_bus.batch but also got an RTIO sequence error.

For example,

with self.dds_bus.batch:
            self.dds0.init()
            self.dds2.init()

Any idea why the delay is necessary?

sbourdeauducq, 2015-08-28:
``dds_bus.batch`` is for sending regular ``set`` commands and cannot be used with ``init``. The delay is necessary because the DDSes are initialized one by one and you need to wait (mostly) for the end of the DAC calibration.

closed 2015-08-28 by sbourdeauducq

raghavendrasrinivas, 2015-08-28:
So the DAC calibration is about 200 us for a 2.4 GHz reference clock. Between the dds_init commands I need a delay of about 250 us which seems reasonable. However, that was the ref_period argument of "core" in ddb.pyon was set to 1ns. If it is set to 10 ns (assuming 100 MHz reference to core device), I need 2500 us between dds_init commands. Why is this the case?

Also, why can't dac calibrations be done in parallel across DDS'?

sbourdeauducq, 2015-08-29:
ref_period should be 1/8 of the clock input on KC705.

Calibrations can be done in parallel but this is slightly more complicated and I see no use for this feature.